riscv/meta-riscv #268: kernel modules fail to insert on qemuriscv32

kraj, 2021-02-12:
build a simple image which has some kernel modules also part of image and then it fails to insert the module.
see the below steps.

```
root@qemuriscv32:~# lsmod
Module                  Size  Used by
root@qemuriscv32:~# modinfo /lib/modules/5.10.12-yocto-standard/kernel/fs/binfmt_misc.ko
filename:       /lib/modules/5.10.12-yocto-standard/kernel/fs/binfmt_misc.ko
license:        GPL
alias:          fs-binfmt_misc
depends:        
intree:         Y
name:           binfmt_misc
vermagic:       5.10.12-yocto-standard SMP preempt mod_unload riscv
root@qemuriscv32:~# modprobe binfmt_misc
modprobe: ERROR: could not insert 'binfmt_misc': Invalid argument
root@qemuriscv32:~# uname -a 
Linux qemuriscv32 5.10.12-yocto-standard #1 SMP PREEMPT Thu Feb 11 20:16:28 UTC 2021 riscv32 riscv32 riscv32 GNU/Linux
root@qemuriscv32:~# zcat /proc/config.gz  | grep MODULE
CONFIG_MODULE_SECTIONS=y
CONFIG_MODULES_USE_ELF_RELA=y
CONFIG_MODULES=y
# CONFIG_MODULE_FORCE_LOAD is not set
CONFIG_MODULE_UNLOAD=y
# CONFIG_MODULE_FORCE_UNLOAD is not set
# CONFIG_MODULE_SRCVERSION_ALL is not set
# CONFIG_MODULE_SIG is not set
# CONFIG_MODULE_COMPRESS is not set
# CONFIG_MODULE_ALLOW_MISSING_NAMESPACE_IMPORTS is not set
CONFIG_MODULES_TREE_LOOKUP=y
root@qemuriscv32:~# 
```

pino-kim, 2021-04-09:
Hi, @kraj long time no see.
Is this issue still alive?

kraj, 2021-04-09:
yes it still fails

```
root@qemuriscv32:~# modprobe nfsd
[  141.976155] nfsd: target 00000000a06c442c can not be addressed by the 32-bit offset from PC = e173c030
modprobe: ERROR: could not insert 'nfsd': Invalid argument

root@qemuriscv32:~# uname -a
Linux qemuriscv32 5.10.25-yocto-standard #1 SMP PREEMPT Fri Apr 9 06:51:13 UTC 2021 riscv32 GNU/Linux
```

pino-kim, 2021-08-10:
Hello @kraj 

I think it is already fixed but I follow your command
And It looks fine.

```
root@qemuriscv32:~# uname -a 
Linux qemuriscv32 5.13.5-yocto-standard #1 SMP PREEMPT Tue Aug 10 14:19:50 UTC 2021 riscv32 riscv32 riscv32 GNU/Linux
root@qemuriscv32:~# modinfo nfsd
filename:       /lib/modules/5.13.5-yocto-standard/kernel/fs/nfsd/nfsd.ko
license:        GPL
author:         Olaf Kirch <okir@monad.swb.de>
alias:          fs-nfsd
depends:        
intree:         Y
name:           nfsd
vermagic:       5.13.5-yocto-standard SMP preempt mod_unload riscv
parm:           cltrack_prog:Path to the nfsdcltrack upcall program (string)
parm:           cltrack_legacy_disable:Disable legacy recoverydir conversion. Default: false (bool)
parm:           nfs4_disable_idmapping:Turn off server's NFSv4 idmapping when using 'sec=sys' (bool)
parm:           inter_copy_offload_enable:Enable inter server to server copy offload. Default: false (bool)
root@qemuriscv32:~# modprobe nfsd
root@qemuriscv32:~# dmesg | grep tail
root@qemuriscv32:~# lsmod
Module                  Size  Used by
sch_fq_codel           24576  1
nfsd                  499712  1

```

kraj, 2026-10-12:
this has been working ok on qemuriscv32 for some time now.

closed 2026-10-12 by kraj