riscv/meta-riscv #276: ERROR: Layer openembedded-layer is not compatible with the core layer which only supports these series: dunfell (layer is compatible with gatesgarth)

Kottoz, 2021-03-25:
i'm trying to build project using Yocto, and this error faced me, how can I solve it ?

```
ERROR: Unable to start bitbake server (None)
ERROR: Server log for this session (/home/kotoz/sys_updr_exp_0-0/rpi64/build/bitbake-cookerdaemon.log):
--- Starting bitbake server pid 16167 at 2021-03-25 11:44:48.669873 ---
ERROR: Layer openembedded-layer is not compatible with the core layer which only supports these series: dunfell (layer is compatible with gatesgarth)
```

Kottoz, 2021-03-25:
Edit the projectDir/layers/local/conf/layer.conf file to add dunfell, for example:

-------

LAYERSERIES_COMPAT_userspace-test = "warrior zeus dunfell gatesgarth"

-------

Save and rebuild.

kraj, 2021-04-07:
hmm how did you. check the layers. It seems you are using master or gatesgrath release of meta-openembedded repo but dunfell for oe-core/poky, you should. be using same branch.

Kottoz, 2021-04-08:
Thanks @kraj , i have switched to the same branch and it's worked.
i will close it.

closed 2021-04-08 by Kottoz