riscv/meta-riscv #278: Bitbake AttributeError

caizixian, 2021-04-11:
**Description**

**Steps to reproduce the issue:**
```
mkdir riscv-yocto-master && cd riscv-yocto-master
repo init -u git://github.com/riscv/meta-riscv  -b master -m tools/manifests/riscv-yocto.xml
repo sync
. meta-riscv/setup.sh
MACHINE=qemuriscv64 bitbake core-image-full-cmdline
```

**Describe the results you received:**
```
ERROR: Command execution failed: Traceback (most recent call last):                                                                                               | ETA:  --:--:--
  File "/home/zixianc/riscv-yocto-master/openembedded-core/bitbake/lib/bb/command.py", line 114, in runAsyncCommand
    self.cooker.updateCache()
  File "/home/zixianc/riscv-yocto-master/openembedded-core/bitbake/lib/bb/cooker.py", line 1621, in updateCache
    self.parser = CookerParser(self, mcfilelist, total_masked)
  File "/home/zixianc/riscv-yocto-master/openembedded-core/bitbake/lib/bb/cooker.py", line 2085, in __init__
    self.bb_caches = bb.cache.MulticonfigCache(self.cfgbuilder, self.cfghash, cooker.caches_array)
  File "/home/zixianc/riscv-yocto-master/openembedded-core/bitbake/lib/bb/cache.py", line 800, in __init__
    loaded += c.prepare_cache(progress)
  File "/home/zixianc/riscv-yocto-master/openembedded-core/bitbake/lib/bb/cache.py", line 406, in prepare_cache
    self.logger.debug("Cache dir: %s", self.cachedir)
  File "/home/zixianc/riscv-yocto-master/openembedded-core/bitbake/lib/bb/__init__.py", line 48, in _debug_helper
    return self.bbdebug(1, *args, **kwargs)
  File "/home/zixianc/riscv-yocto-master/openembedded-core/bitbake/lib/bb/__init__.py", line 59, in bbdebug
    if self.name in bb.msg.loggerDefaultDomains and loglevel > (bb.msg.loggerDefaultDomains[self.name]):
AttributeError: 'PrefixLoggerAdapter' object has no attribute 'name'
```

**Describe the results you expected:**
Builds the image successfully.

**Additional information you deem important (e.g. issue happens only occasionally):**
If I checkout all layers to be dunfell and checkout `openembedded-core/bitbake` to be `2020-04.6-dunfell`, the image building works as expected.

**Additional details (revisions used, host distro, etc.):**
Host: Ubuntu 18.04.5 LTS with 5.4.0-64-generic kernel.

kraj, 2021-04-11:
Can you try it with Yoe Distro ?
https://github.com/yoedistro/yoe-distro

and report if you still can reproduce it ?

caizixian, 2021-04-15:
Should I use https://github.com/YoeDistro/yoe-distro/blob/master/qemuriscv64-envsetup.sh

kraj, 2021-04-15:
@caizixian yes thats the right one for rv64

kraj, 2022-03-24:
I am closing this