riscv/meta-riscv #280: Document beagleV bringup steps

kraj, 2021-04-30:
We need to document how to unfold beagleV step by step and build an image which can be booted on beagleV, perhaps starting from when one received the board shipment.

pdp7, 2021-04-30:
I am trying to add resources here:
https://github.com/beagleboard/beaglev-starlight/blob/main/README.md

Please let me know what I can provide to help the documentation effort

@tpetazzoni buildroot branch has good details that might help oe/yocto:
https://github.com/tpetazzoni/buildroot/tree/beaglev

kraj, 2021-04-30:
@pdp7 thanks thats a good start we probably can add a `beagleV.md` file on the topdir and start with steps.

macpijan, 2021-05-04:
Added some basic readme to starti with in: https://github.com/riscv/meta-riscv/pull/279/files#diff-4a464ffb2af727f2accca11adf3ecbf92cfc51c0743ba66b144b091fcff9fc66

kraj, 2021-05-07:
we now have merged initial support. Which includes documentation on bringing up beagleV as well.

roemvaar, 2021-05-08:
I'm currently working on a tutorial for the BeagleV Starlight board bring up - If you want, I can make sure that this material gets merged into the docs.

kraj, 2021-05-08:
soungs good @roemvaar

pino-kim, 2021-08-05:
This Is not a proper site to ask about beagleV But.
beagleV is postponed or canceled?

alistair23, 2021-08-05:
You will have to look at the BeagleV communiyt: https://forum.beagleboard.org/t/the-future-of-beaglev-community/30463

closed 2021-08-05 by alistair23

pdp7, 2021-08-05:
@pino-kim feel free to either ask questions on my forum post that was linked above or through email: drew@beagleboard.org