riscv/meta-riscv #302: Old override syntax error

sdanfa, 2021-08-02:
Hello,
I'm trying to use instructions from this git repository to build with yocto but I'm getting errors. Running setup.sh gives errors:
`Adding layers
NOTE: Starting bitbake server...
ERROR: Variable NON_MULTILIB_RECIPES_append contains an operation using the old override syntax. Please convert this layer/metadata before attempting to use with a newer bitbake.
ERROR: Parse failure with the specified layer added, aborting.
NOTE: Starting bitbake server...
ERROR: Layer meta-python is not compatible with the core layer which only supports these series: honister (layer is compatible with hardknott)
ERROR: Parse failure with the specified layer added, aborting.
NOTE: Starting bitbake server...
ERROR: Layer multimedia-layer is not compatible with the core layer which only supports these series: honister (layer is compatible with hardknott)
ERROR: Parse failure with the specified layer added, aborting.
NOTE: Starting bitbake server...
ERROR: Layer networking-layer is not compatible with the core layer which only supports these series: honister (layer is compatible with hardknott)
ERROR: Parse failure with the specified layer added, aborting.
NOTE: Starting bitbake server...
ERROR: Variable INSANE_SKIP_append_pn-eudev_riscv64 contains an operation using the old override syntax. Please convert this layer/metadata before attempting to use with a newer bitbake.
ERROR: Parse failure with the specified layer added, aborting.
Creating auto.conf
To build an image run
---------------------------------------------------
MACHINE=qemuriscv64 bitbake core-image-full-cmdline
---------------------------------------------------

Buildable machine info
---------------------------------------------------
* qemuriscv64: The 64-bit RISC-V machine
* qemuriscv32: The 32-bit RISC-V machine
* freedom-u540: The SiFive HiFive Unleashed board
---------------------------------------------------
`

kraj, 2021-08-02:
@sdanfa you must be using master-next branch of poky/oe-core repository, we do have changes needed for working with new override syntax but they are not yet proposed to meta-riscv see https://github.com/YoeDistro/meta-riscv/tree/yoe/mut

sdanfa, 2021-08-02:
> @sdanfa you must be using master-next branch of poky/oe-core repository, we do have changes needed for working with new override syntax but they are not yet proposed to meta-riscv see https://github.com/YoeDistro/meta-riscv/tree/yoe/mut

HI thanks for the quick reply.
I'm new to yocto, can you please tell me where and what to change to make it use master-next branch of poky/oe-core please?

Thanks!

sdanfa, 2021-08-02:
@kraj I tried using your new repo but I'm getting same errors. I used your repo using command `repo init -u git://github.com/YoeDistro/meta-riscv -b yoe/mut -m tools/manifests/riscv-yocto.xml`

sdanfa, 2021-08-02:
@kraj I've tried changing manifests file oe-core to master-next too but same errors. `  <project name="openembedded-core" remote="openembedded" path="openembedded-core" revision="master-next" />
`

kraj, 2022-03-24:
@sdanfa  is this issue still persisting ?

abusous2000, 2022-12-28:
You need to use 
`IMAGE_INSTALL:append`
instead of
`IMAGE_INSTALL_append`
and you will be OK

ZhaoqiWang1921, 2024-01-16:
> You need to use
> `IMAGE_INSTALL:append`
> instead of
> `IMAGE_INSTALL_append`
> and you will be OK

> You need to use
> `IMAGE_INSTALL:append`
> instead of
> `IMAGE_INSTALL_append`
> and you will be OK

Thank you, this is very useful!

ZhaoqiWang1921, 2024-01-16:
@abusous2000 Thank you, this is very useful!

kraj, 2024-06-14:
I am closing it since it seems @sdanfa found way through it. If it still is persisting please re-open.

closed 2024-06-14 by kraj