riscv/meta-riscv #314: [Question] Honister branch support

alleriino, 2022-02-09:
**Question:**

Hi Khem, able to ask if currently are there any plans to have support for the honister branch? 

This is because that I noticed there are some minor discrepancies between the master and honister branch, 
which may cause some issues in future if people are using the latest oe-core honister branch with the meta-riscv
master branch (for example the busybox version of 1.35 in master vs 1.34 in honister). Thus, I was wondering if honister
branch will be supported in the future.

Thanks!

kraj, 2022-02-09:
@alleriino yes, thanks for bringing it to my notice. I think we should branch honister at this point.

kraj, 2022-02-09:
branch should be created now.

closed 2022-02-09 by kraj