riscv/meta-riscv #336: Multilib support

lapnd, 2022-04-25:
Hi,
Using riscv-gnu-toolchain, we will able to get pre-compiled system libraries for all ABIs by using option `--enable-multilib`
Do we support this kind of features?

kraj, 2022-04-25:
@lapnd, the multilib concept in OE/Yocto is a bit different., since its building systems, we usually pick one multilib per toolchain.

lapnd, 2022-04-25:
If I understand correctly, let's say, If I want to support `rv32i/ilp32, rv32iac/ilp32, rv32im/ilp32, rv32imac/ilp32, rv32imafc/ilp32f, rv32imafdc/ilp32d, rv64imac/lp64, rv64imafdc/lp64d` I have to build 08 toolchains /08 SDKs. Is this correct?
Thank you @kraj

kraj, 2022-04-25:
for linux yes. For baremetal this might be doable to build a single SDK containing all needed runtime with some changes.

lapnd, 2022-04-25:
Could you please give me few hints on what to change to have single
baremeral sdk for all target?
Thank you

closed 2022-10-11 by lapnd