riscv/meta-riscv #345: Pie is not supported

Weupgroup, 2022-07-03:
Hi,
I’m trying to use meta-riscv with Poky. The build is fine. Howerver, when using elf toolchain,the pie is not supported as 
```sh

Building target: HelloworldDemo.elf
Invoking: GNU RISC-V Cross C Linker
riscv64-poky-elf-gcc -mcmodel=medlow -msmall-data-limit=8 -mno-save-restore -march=rv64imac_zicsr_zifencei -mabi=lp64 -O0 -w  -g -T "" -nostartfiles -Xlinker --gc-sections -Wl,-Map,"HelloworldDemo.map" -o "HelloworldDemo.elf"  ./src/target-specific/libfemto/std/abort.o ./src/target-specific/libfemto/std/clz.o ./src/target-specific/libfemto/std/ctz.o ./src/target-specific/libfemto/std/exit.o ./src/target-specific/libfemto/std/getchar.o ./src/target-specific/libfemto/std/malloc.o ./src/target-specific/libfemto/std/memchr.o ./src/target-specific/libfemto/std/memcmp.o ./src/target-specific/libfemto/std/memcpy.o ./src/target-specific/libfemto/std/memset.o ./src/target-specific/libfemto/std/printf.o ./src/target-specific/libfemto/std/purtchar.o ./src/target-specific/libfemto/std/puts.o ./src/target-specific/libfemto/std/snprintf.o ./src/target-specific/libfemto/std/strchr.o ./src/target-specific/libfemto/std/strcmp.o ./src/target-specific/libfemto/std/strlen.o ./src/target-specific/libfemto/std/strncmp.o ./src/target-specific/libfemto/std/strncpy.o ./src/target-specific/libfemto/std/vprintf.o ./src/target-specific/libfemto/std/vsnprintf.o  ./src/target-specific/libfemto/drivers/htif.o ./src/target-specific/libfemto/drivers/neorv32_uart.o ./src/target-specific/libfemto/drivers/ns16550a.o ./src/target-specific/libfemto/drivers/semihost.o  ./src/target-specific/libfemto/arch/riscv/auxval.o ./src/target-specific/libfemto/arch/riscv/csr.o ./src/target-specific/libfemto/arch/riscv/device.o ./src/target-specific/libfemto/arch/riscv/memory.o ./src/target-specific/libfemto/arch/riscv/pmp.o ./src/target-specific/libfemto/arch/riscv/spinlock.o ./src/target-specific/libfemto/arch/riscv/start.o ./src/target-specific/libfemto/arch/riscv/trap.o  ./src/environmental-support/virt/crt.o ./src/environmental-support/virt/setup.o  ./src/environmental-support/common/rv32/macros.o  ./src/environmental-support/common/constants.o  ./src/HelloworldDemo.o   
~/toolchain/riscv64-poky-elf/sysroots/x86_64-pokysdk-linux/usr/libexec/riscv64-poky-elf/gcc/riscv64-poky-elf/11.2.0/real-ld: -pie not supported
collect2: error: ld returned 1 exit status
make: *** [HelloworldDemo.elf] Error 1
makefile:56: recipe for target 'HelloworldDemo.elf' failed
```

Can you please give me few hints? 
I’m just a newbie with poky and meta-riscv.
Thank you

kraj, 2024-06-14:
@chipsme I think the problem perhaps is  in `meta/conf/distro/include/security_flags.inc`
```
GCCPIE ?= "--enable-default-pie"
```
perhaps we should define it to empty string in `conf/machine/ baremetal-riscv64.conf`

kraj, 2026-10-12:
closing for lack of activity

closed 2026-10-12 by kraj