riscv/meta-riscv #355: Fetch error for starfive-tech u-boot

pratikfarkase94, 2022-08-17:
**Description**

**Describe the results you received:**

More info in the attached screenshots. 
[image]

kraj, 2022-08-17:
@pratikfarkase94 describe your setup a little bit more. Secondly use latest master branch, here is what I did

```
% MACHINE=visionfive bitbake -e virtual/bootloader | grep SRCREV=
SRCREV="ccecef294d355e9d05edf0bb6058002a0fe08908"

```

which is different than what you reported

pratikfarkase94, 2022-08-18:
SRCREV="ccecef294d355e9d05edf0bb6058002a0fe08908" as per "MACHINE=visionfive bitbake -e virtual/bootloader | grep SRCREV=SRCREV="ccecef294d355e9d05edf0bb6058002a0fe08908"" . 

Branch under use is master. Steps i followed : 

mkdir riscv-yocto && cd riscv-yocto
repo init -u https://github.com/riscv/meta-riscv  -b master -m tools/manifests/riscv-yocto.xml
repo sync
repo start work --all
. ./meta-riscv/setup.sh
MACHINE=beaglev-starlight-jh7100 bitbake core-image-weston

That's when i hit this error. I tried using a different internet connection but the issue seems to be same.

kraj, 2022-08-18:
I see, so you are building for beagleV where I think the SRCREV might be stale and we have to upgrade that to something valid branch + srcrev but I think MACHINE=visionfive should work out fine. Can you try out this patch

https://github.com/riscv/meta-riscv/pull/356

and report back ?

pratikfarkase94, 2022-08-18:
Works! Thanks. closing now.

closed 2022-08-18 by pratikfarkase94