riscv/meta-riscv #357: linux-starfive-dev/5.19.0-rc4+ kernel compilation fails on beaglev master branch

pratikfarkase94, 2022-08-18:
**Description**

**Steps to reproduce the issue:**
1.
2.
3.

**Describe the results you received:**

**Describe the results you expected:**

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

pratikfarkase94, 2022-08-18:
[kernel_compile_fail.txt](https://github.com/riscv/meta-riscv/files/9371369/kernel_compile_fail.txt)

kraj, 2022-08-18:
try https://github.com/riscv/meta-riscv/pull/356/commits/d3754363b73482c2804442ba494bd71ccf90b2a5 and report back

pratikfarkase94, 2022-08-18:
[kernel_compile_fail.txt](https://github.com/riscv/meta-riscv/files/9376179/kernel_compile_fail.txt)
more compilation errors.

PR #470 (2024-05-06): beaglev-starlight: fix u-boot and kernel compile
fixes #357 and #358 

Error output when building u-boot-starfive_v2021.04 from the master branch:

```
| scripts/kconfig/conf  --defconfig=arch/../configs/starfive_jh7100_beagle_v_smode_defconfig Kconfig
| ***
| *** Can't find default configuration "arch/../configs/starfive_jh7100_beagle_v_smode_defconfig"!
| ***
| make[2]: *** [/opt/riscv-yocto/build/tmp/work/beaglev_starlight_jh7100-poky-linux/u-boot-starfive/v2021.04/git/scripts/kconfig/Makefile:110: starfive_jh7100_beagle_v_smode_defconfig] Error 1
| make[1]: Leaving directory '/opt/riscv-yo [... cut]
closed 2024-05-06 by kraj