riscv/meta-riscv #364: do_rootfs error

zhangze0722, 2022-09-16:
**Description**

**Steps to reproduce the issue:**
1.download the newest code
   repo init -u https://github.com/riscv/meta-riscv  -b master -m tools/manifests/riscv-yocto.xml
   repo sync
   repo start work --all

openembedded-core version:
commit f9c23404de44553eacd363885588b88714742387 (HEAD -> work, openembedded/master, m/master)
Author: Khem Raj <raj.khem@gmail.com>
Date:   Mon Sep 12 18:37:45 2022 -0700

    inetutils: Fix remote DoS vulnerability in inetutils-telnetd
    
    Signed-off-by: Khem Raj <raj.khem@gmail.com>
    Signed-off-by: Richard Purdie <richard.purdie@linuxfoundation.org>

2.source ./meta-riscv/setup.sh
3.MACHINE=qemuriscv64 bitbake core-image-full-cmdline

**Describe the results you received:**

ERROR: The postinstall intercept hook 'update_font_cache' failed,

**Describe the results you expected:**

create image OK

**Additional information you deem important (e.g. issue happens only occasionally):**

create packages dir to avoid update_mime_database error

diff --git a/scripts/postinst-intercepts/update_mime_database b/scripts/postinst-intercepts/update_mime_database
index 582d1e162c..77f3d19b54 100644
--- a/scripts/postinst-intercepts/update_mime_database
+++ b/scripts/postinst-intercepts/update_mime_database
@@ -5,5 +5,18 @@
 # Post-install intercept for mime.bbclass
 
 echo "Updating MIME database... this may take a while."
+mkdir -p $D${mimedir}/packages
 update-mime-database $D${mimedir}

**Additional details (revisions used, host distro, etc.):**

ubuntu 20.04 x86_64

kraj, 2022-09-16:
this could be due to qemu usermode crash.

zhangze0722, 2022-09-17:
not runqemu yet, cannot create image correctly.

kraj, 2022-09-17:
yeah qemu is also used durning build in usermode to run some target binaries. Can you find out the failing script which is running update_font_cache and inspect it for any qemu crashes.

zhangze0722, 2022-09-20:
It seems no difference between this version and previous version.
skip this step by changing this file:
zhangze@zhangze-virtual-machine:~/riscv-yocto.0919/openembedded-core$ git diff 
diff --git a/meta/lib/oe/rootfs.py b/meta/lib/oe/rootfs.py
index 8587c5db82..af7e415154 100644
--- a/meta/lib/oe/rootfs.py
+++ b/meta/lib/oe/rootfs.py
@@ -207,7 +207,7 @@ class Rootfs(object, metaclass=ABCMeta):
 
         execute_pre_post_process(self.d, rootfs_post_install_cmds)
 
-        self.pm.run_intercepts()
+        #self.pm.run_intercepts()
 
         execute_pre_post_process(self.d, post_process_cmds)

maybe because qemu upgrade from 7.0 to 7.1 ??

kraj, 2022-09-20:
It’s possible perhaps try reverting this commit to upgrade qemu and see if this helps

kraj, 2022-10-11:
@zhangze0722 are you still stuck at this issue ?

zhangze0722, 2022-10-11:
cannot be reproduced.

closed 2022-10-11 by zhangze0722