riscv/meta-riscv #366: cannot enable gfortran in core-image-full-cmdline

zhangze0722, 2022-09-30:
**Description**
cannot enable gfortran in core-image-full-cmdline

**Steps to reproduce the issue:**
1.
~/riscv-yocto/meta-rvcore/recipes-devtools/gcc$ cat gcc-runtime_%.bbappend 
#FILESEXTRAPATHS:prepend := "${THISDIR}/${PN}:"
RUNTIMETARGET += "libgfortran"

2.
~/riscv-yocto/build$ cat conf/local.conf
#enable fortran
FORTRAN_forcevariable = ",fortran"
#FORTRAN = ",fortran"

3.

**Describe the results you received:**

| checking for cos in -lm... yes
| checking for riscv64-unknown-linux-gfortran... (cached) no
| checking whether we are using the GNU Fortran compiler... (cached) no
| checking whether no accepts -g... (cached) no
| checking whether the GNU Fortran compiler is working... no
| configure: error: GNU Fortran is not working; please report a bug in http://gcc.gnu.org/bugzilla, attaching /home/zhangze/riscv-yocto/build/tmp-glibc/work/riscv64-unknown-linux/gcc-runtime/12.2.0-r0/gcc-12.2.0/build.riscv64-unknown-linux.riscv64-unknown-linux/riscv64-unknown-linux/libgfortran/config.log
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/zhangze/riscv-yocto/openembedded-core/meta/recipes-devtools/gcc/gcc-runtime_12.2.bb:do_configure) failed with exit code '1'

**Describe the results you expected:**

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

zhangze0722, 2022-09-30:
FORTRAN_TOOLS = " \
    gfortran \
    gfortran-symlinks \
    libgfortran \
    libgfortran-dev \
"
IMAGE_INSTALL:append = " openjdk \
                        openjdk-demo \
                        openjdk-doc \
                        openjdk-dev \
                        e2fsprogs \
                        e2fsprogs-resize2fs \
                        python3-pip \
                        gcc gcc-symlinks gcc-plugins \
                        gcc-dev \
                        gcc-doc \
                        gcov gcov-symlinks \
                        g++ g++-symlinks\
                        cpp cpp-symlinks \
                        binutils \
                        go \
                        go-helloworld \
                        vim \
                        time \
[config.log](https://github.com/riscv/meta-riscv/files/9682599/config.log)
                        ${FORTRAN_TOOLS}  \
                        perl \
"

zhangze0722, 2022-09-30:
~/riscv-yocto/build$ bitbake-layers show-recipes |grep fortran
libgfortran:
  meta                 12.2 (skipped: libgfortran needs fortran support to be enabled in the compiler)

zhangze0722, 2022-09-30:
~/riscv-yocto/build$ bitbake libgfortran
Loading cache: 100% |#######################################################################################################################################################################| Time: 0:00:00
Loaded 4013 entries from dependency cache.
ERROR: Nothing PROVIDES 'libgfortran'
libgfortran was skipped: libgfortran needs fortran support to be enabled in the compiler

Summary: There was 1 ERROR message, returning a non-zero exit code.

zhangze0722, 2022-09-30:
Build Configuration:
BB_VERSION           = "2.0.1"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv64-unknown-linux"
MACHINE              = "qemuriscv64"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv64"
meta                 = "work:fb1de2abc53bd742bc55cfecd384b78852c10d80"
meta-oe              
meta-python          
meta-multimedia      
meta-networking      = "work:de66eb0c0dae0930f9e1ba7a358db1ae6b3f2849"
meta-riscv           = "work:c5f397f37eadb1209ec39202b48a7592733aabd4"
meta-rvcore          = "<unknown>:<unknown>"

kraj, 2022-09-30:
Can you try

```
FORTRAN:forcevariable = ",fortran"

```

zhangze0722, 2022-10-02:
bitbake-layers show-recipes *fortran*
NOTE: Starting bitbake server...
Loading cache: 100% |##################################################################################################################################| Time: 0:00:00
Loaded 4014 entries from dependency cache.
=== Matching recipes: ===
libgfortran:
  meta                 12.2.0

but, still occur this error:

| checking for cos in -lm... yes
| checking for riscv64-unknown-linux-gfortran... (cached) no
| checking whether we are using the GNU Fortran compiler... (cached) no
| checking whether no accepts -g... (cached) no
| checking whether the GNU Fortran compiler is working... no
| configure: error: GNU Fortran is not working; please report a bug in http://gcc.gnu.org/bugzilla, attaching /media/zhangze/4TB/riscv-yocto/build/tmp-glibc/work/riscv64-unknown-linux/gcc-runtime/12.2.0-r0/gcc-12.2.0/build.riscv64-unknown-linux.riscv64-unknown-linux/riscv64-unknown-linux/libgfortran/config.log

zhangze0722, 2022-10-02:
this message disappeared:
libgfortran was skipped: libgfortran needs fortran support to be enabled in the compiler

but has configure error, can you explain the principle behind this?
i am bitbake gcc-runtime or libgfortran, why it check "whether the GNU Fortran compiler is working" ?
i want to compile fortran compiler

kraj, 2022-10-11:
@zhangze0722 fortran support comes from a separate recipe called `libgfortran`, Please try setting

```
FORTRAN:forcevariable = ",fortran"
```
in `local.conf` and try building again.

zhangze0722, 2022-10-11:
thanks a lot, I found that if no gcc-runtime_%.bbappend, gfortran will be ok.
1.
~/riscv-yocto/meta-rvcore/recipes-devtools/gcc$ cat gcc-runtime_%.bbappend
#FILESEXTRAPATHS:prepend := "${THISDIR}/${PN}:"
RUNTIMETARGET += "libgfortran"

closed 2022-10-11 by zhangze0722