riscv/meta-riscv #374: perl-misc do_rootfs error

zhangze0722, 2022-11-14:
**Description**

Loading cache: 100% |######################################################################################################################################################################| Time: 0:00:00
Loaded 4573 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.2.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv64-unknown-linux"
MACHINE              = "qemuriscv64"
DISTRO               = "rvcore-distro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv64"
meta                 = "HEAD:d801d866e1d70abe229d17a637636634f81b87e3"
meta-oe              
meta-python          
meta-multimedia      
meta-networking      = "HEAD:d04444509a220fcb61496d7e64f3ba09c647543b"
meta-riscv           = "HEAD:975227ea8468409aaf75860f45b03d8895140859"
meta-rvcore          = "master:f5f22eaf0953aceb350ba2dccb71dddb1c89d924"
meta-clang           = "master:e8fd396155e1f8a5cdb57e7abd0c1eb698635371"
meta-perl            
meta-filesystems     = "HEAD:d04444509a220fcb61496d7e64f3ba09c647543b"
meta-selinux         = "master:506daf988c844e857c674ef70a46476a005bf17b"
meta-security        = "master:2aa48e6f4e519abc7d6bd56da2c067309a303e80"
meta-virtualization  = "master:9f41aca8e6cd31ba2752dba5d375cf19009a6ec3"
meta-cloud-services  = "master:83c836dc5dd39fbd0f661afabc56449d7caf1cd4"
meta-cgl-common      = "master:64b8d64a7c4d8ee92bd9857f1bfaa67db699f743"

Initialising tasks: 100% |#################################################################################################################################################################| Time: 0:00:06
Sstate summary: Wanted 2 Local 0 Mirrors 0 Missed 2 Current 2400 (0% match, 99% complete)
NOTE: Executing Tasks
WARNING: rvcore-image-full-cmdline-1.0-r0 do_rootfs: perl-misc.postinst returned 1, marking as unpacked only, configuration required on target.
### ERROR: rvcore-image-full-cmdline-1.0-r0 do_rootfs: Postinstall scriptlets of ['perl-misc'] have failed. If the intention is to defer them to first boot,
then please place them into pkg_postinst_ontarget:${PN} ().
Deferring to first boot via 'exit 1' is no longer supported.
Details of the failure are in /home/zhangze/rvcore_yocto/build/tmp-glibc/work/qemuriscv64-unknown-linux/rvcore-image-full-cmdline/1.0-r0/temp/log.do_rootfs.
ERROR: Logfile of failure stored in: /home/zhangze/rvcore_yocto/build/tmp-glibc/work/qemuriscv64-unknown-linux/rvcore-image-full-cmdline/1.0-r0/temp/log.do_rootfs.4156760
ERROR: Task (/home/zhangze/rvcore_yocto/meta-rvcore/recipes-extended/images/rvcore-image-full-cmdline.bb:do_rootfs) failed with exit code '1'
NOTE: Tasks Summary: Attempted 5988 tasks of which 5987 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 4 seconds

Summary: 1 task failed:
  /home/zhangze/rvcore_yocto/meta-rvcore/recipes-extended/images/rvcore-image-full-cmdline.bb:do_rootfs
Summary: There was 1 WARNING message.
Summary: There was 1 ERROR message, returning a non-zero exit code.

**Steps to reproduce the issue:**
1. In image bb file, IMAGE_INSTALL add perl-misc, 
2. when I run bitbake core-image-full-cmdline, perl-misc do_rootfs error
3. software is the newest meta-riscv layer

**Describe the results you received:**

**Describe the results you expected:**

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

zhangze0722, 2022-11-14:
maybe conflict with
meta-openembedded/meta-perl/recipes-perl/libio/libio-compress-perl_2.201.bb

kraj, 2022-11-14:
post installs fails when sometimes qemu segfaults in cross builds when running some target operations during build. Can you find details in logs as to why it thinks it failed ?

zhangze0722, 2023-01-16:
disappear

zhangze0722, 2023-01-29:
this issue is due to two perl packages conflict:
../meta-openembedded/meta-perl/recipes-perl/libio/libio-compress-perl_2.201.bb  produces libio-compress-perl.ipk
openembedded-core/meta/recipes-devtools/perl/perl_5.36.0.bb  produces perl-misc.ipk

both provide /usr/bin/streamzip and /usr/bin/zipdetails

in libio-compress-perl_2.%.bbappend, add these will fix this:

FILESEXTRAPATHS:prepend := "${THISDIR}/${PN}:"
#FILES:${PN}:remove = "${bindir}/streamzip"

do_install:append() {

    rm ${D}${bindir}/streamzip
    rm ${D}${bindir}/zipdetails
    rmdir ${D}${bindir}
}

very strange, this line doesn't work, so I have to rm files in do_install(), maybe it's a bug
>  FILES:${PN}:remove = "${bindir}/streamzip"

kraj, 2024-06-14:
@zhangze0722 I think you have found a packaging bug here. We need to use `update-alternatives` mechanism here and let one of the two packages install it based on priority

https://docs.yoctoproject.org/dev/singleindex.html#ref-classes-update-alternatives

closed 2024-06-15 by kraj