riscv/meta-riscv #384: gawk's -ptest package has QA issue

zhangze0722, 2023-01-17:
**Description**

gawk's -ptest package has QA issue

**Steps to reproduce the issue:**
1.source meta-riscv/setup.sh
2.bitbake core-image-full-cmdline
3.gawk-test error occour

**Describe the results you received:**
BB_VERSION           = "2.2.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv64-unknown-linux"
MACHINE              = "qemuriscv64"
DISTRO               = "rvcore-distro"
DISTRO_VERSION       = "1.1"
TUNE_FEATURES        = "riscv64"
meta                 = "work:9b0f88642570383b09de94238fe1a8ef5119fb47"
meta-oe              
meta-python          
meta-multimedia      
meta-networking      = "work:ca7082caa800124b4e0a7b36175f60387d1e3f8c"
meta-riscv           = "work:d6e3efd54a3c1361fecf2a56c6f4f590fbe676d9"
meta-clang           = "master:57a06dffa2d528456fa973592dc1989cb1342f29"
meta-rvcore          = "main:96183eb2029c01afbaafc68cd3de8f139212b44e"
meta-perl            
meta-filesystems     = "work:ca7082caa800124b4e0a7b36175f60387d1e3f8c"

Initialising tasks: 100% |############################################################################################################################################################################################################| Time: 0:00:01
Sstate summary: Wanted 398 Local 0 Mirrors 0 Missed 398 Current 2072 (0% match, 83% complete)
NOTE: Executing Tasks
ERROR: gawk-5.2.1-r0 do_package_qa: QA Issue: /usr/lib/gawk/ptest/test/fnarydel.awk contained in package gawk-ptest requires /usr/local/bin/gawk, but no providers found in RDEPENDS:gawk-ptest? [file-rdeps]
ERROR: gawk-5.2.1-r0 do_package_qa: QA Issue: /usr/lib/gawk/ptest/test/mpfrfield.awk contained in package gawk-ptest requires /bin/gawk, but no providers found in RDEPENDS:gawk-ptest? [file-rdeps]
ERROR: gawk-5.2.1-r0 do_package_qa: Fatal QA errors were found, failing task.
ERROR: Logfile of failure stored in: /home/work2TB/zhangze/rvcore-yocto/build/tmp-glibc/work/riscv64-unknown-linux/gawk/5.2.1-r0/temp/log.do_package_qa.3719989
ERROR: Task (/home/work2TB/zhangze/rvcore-yocto/openembedded-core/meta/recipes-extended/gawk/gawk_5.2.1.bb:do_package_qa) failed with exit code '1'

**Describe the results you expected:**

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

alistair23, 2023-01-17:
This seems like an upstream issue

cg-sat, 2024-06-10:
I had the same problem and found the following workaround (if you also need to stay on a specific revision of yocto itself).
In the file `openembedded-core/meta/recipes-extended/gawk/gawk_5.2.1.bb` add the following parameter to the very long `sed` line:
```
-e 's|#! /bin/gawk|#! ${bindir}/gawk|g' \
```

This fixes the hashbang line of the https://git.savannah.gnu.org/cgit/gawk.git/tree/test/mpfrfield.awk file, which still contains the hashbang line today.

kraj, 2024-06-14:
@cg-sat yeah, I think the problem here, I have sent a fix to oe-core layer [1]

[1] https://lore.kernel.org/openembedded-core/20240614190322.726408-1-raj.khem@gmail.com/T/#u

kraj, 2026-10-12:
it seems to be working now a days.

closed 2026-10-12 by kraj