riscv/meta-riscv #385: pulseaudio need bash, not only bash-completion

zhangze0722, 2023-01-17:
**Description**

pulseaudio need bash, not only bash-completion

**Steps to reproduce the issue:**
1.source meta-riscv/setup.sh
2.bitbake core-image-full-cmdline
3.if add this line in pulseaudio_xxx.bb, this error disappear
RDEPENDS:${PN}-bash-completion += "bash"

**Describe the results you received:**

Build Configuration:
BB_VERSION           = "2.2.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv64-unknown-linux"
MACHINE              = "qemuriscv64"
DISTRO               = "rvcore-distro"
DISTRO_VERSION       = "1.1"
TUNE_FEATURES        = "riscv64"
meta                 = "work:9b0f88642570383b09de94238fe1a8ef5119fb47"
meta-oe              
meta-python          
meta-multimedia      
meta-networking      = "work:ca7082caa800124b4e0a7b36175f60387d1e3f8c"
meta-riscv           = "work:d6e3efd54a3c1361fecf2a56c6f4f590fbe676d9"
meta-clang           = "master:57a06dffa2d528456fa973592dc1989cb1342f29"
meta-rvcore          = "main:96183eb2029c01afbaafc68cd3de8f139212b44e"
meta-perl            
meta-filesystems     = "work:ca7082caa800124b4e0a7b36175f60387d1e3f8c"

Initialising tasks: 100% |############################################################################################################################################################################################################| Time: 0:00:01
Sstate summary: Wanted 59 Local 0 Mirrors 0 Missed 59 Current 2348 (0% match, 97% complete)
NOTE: Executing Tasks
WARNING: Variable key RDEPENDS:${PN}-bash-completion (bash-completion) replaces original key RDEPENDS:pulseaudio-bash-completion (bash).
ERROR: pulseaudio-16.1-r0 do_package_qa: QA Issue: /usr/share/bash-completion/completions/pactl contained in package pulseaudio-bash-completion requires /bin/bash, but no providers found in RDEPENDS:pulseaudio-bash-completion? [file-rdeps]
ERROR: pulseaudio-16.1-r0 do_package_qa: Fatal QA errors were found, failing task.
ERROR: Logfile of failure stored in: /home/work2TB/zhangze/rvcore-yocto/build/tmp-glibc/work/riscv64-unknown-linux/pulseaudio/16.1-r0/temp/log.do_package_qa.1217609
ERROR: Task (/home/work2TB/zhangze/rvcore-yocto/openembedded-core/meta/recipes-multimedia/pulseaudio/pulseaudio_16.1.bb:do_package_qa) failed with exit code '1'

**Describe the results you expected:**

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

alistair23, 2023-01-17:
We don't have pulseaudio in meta-riscv though

kraj, 2024-06-14:
https://git.yoctoproject.org/poky/tree/meta/classes-recipe/bash-completion.bbclass#n13

already adds the needed rdep and pulseaudio recipe does inherit bash-completion
https://git.yoctoproject.org/poky/tree/meta/recipes-multimedia/pulseaudio/pulseaudio.inc#n68

not sure how to reproduce it.

kraj, 2026-10-12:
no updates, closing for lack of activity

closed 2026-10-12 by kraj