riscv/meta-riscv #398: Qemu RISCV64 build fails

disdi, 2023-03-12:
**Building console-only image for the 64-bit QEMU machine fails**

**Steps to reproduce the issue:**
 ~/riscv-yocto$ MACHINE=qemuriscv64 bitbake core-image-full-cmdline

`| cc1: some warnings being treated as errors
| Makefile:472: recipe for target 'mount_clnt.o' failed
| make[3]: *** [mount_clnt.o] Error 1
| make[3]: *** Waiting for unfinished jobs....
| make[3]: Leaving directory '/lhome/saksinh/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/nfs-utils/2.6.2-r0/nfs-utils-2.6.2/support/export'
| ERROR: oe_runmake failed
| Makefile:401: recipe for target 'all' failed
| make[2]: Leaving directory '/lhome/saksinh/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/nfs-utils/2.6.2-r0/nfs-utils-2.6.2/support/export'
| Makefile:427: recipe for target 'all-recursive' failed
| make[1]: Leaving directory '/lhome/saksinh/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/nfs-utils/2.6.2-r0/nfs-utils-2.6.2/support'
| Makefile:467: recipe for target 'all-recursive' failed
| WARNING: /lhome/saksinh/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/nfs-utils/2.6.2-r0/temp/run.do_compile.2481686:183 exit 1 from 'exit 1'
| WARNING: Backtrace (BB generated script):
| make[2]: *** [all] Error 2
| make[1]: *** [all-recursive] Error 1
| make: *** [all-recursive] Error 1
| 	#1: bbfatal_log, /lhome/saksinh/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/nfs-utils/2.6.2-r0/temp/run.do_compile.2481686, line 183
| 	#2: die, /lhome/saksinh/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/nfs-utils/2.6.2-r0/temp/run.do_compile.2481686, line 167
| 	#3: oe_runmake, /lhome/saksinh/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/nfs-utils/2.6.2-r0/temp/run.do_compile.2481686, line 162
| 	#4: autotools_do_compile, /lhome/saksinh/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/nfs-utils/2.6.2-r0/temp/run.do_compile.2481686, line 157
| 	#5: do_compile, /lhome/saksinh/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/nfs-utils/2.6.2-r0/temp/run.do_compile.2481686, line 152
| 	#6: main, /lhome/saksinh/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/nfs-utils/2.6.2-r0/temp/run.do_compile.2481686, line 196
ERROR: Task (/lhome/saksinh/riscv-yocto/openembedded-core/meta/recipes-connectivity/nfs-utils/nfs-utils_2.6.2.bb:do_compile) failed with exit code '1'

`

kraj, 2023-03-12:
Can you check for real error message in /lhome/saksinh/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/nfs-utils/2.6.2-r0/temp/run.do_compile.2481686 and report that here ?

disdi, 2023-03-12:
Attaching file .. [run.do_compile.2481686.txt](https://github.com/riscv/meta-riscv/files/10950121/run.do_compile.2481686.txt)

:~$ cat /lhome/saksinh/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/nfs-utils/2.6.2-r0/temp/run.do_compile.2481686 | tail
cd '/lhome/saksinh/riscv-yocto/build/tmp-glibc/work/riscv64-oe-linux/nfs-utils/2.6.2-r0/nfs-utils-2.6.2'
do_compile

cleanup
ret=$?
trap '' 0
exit $ret

kraj, 2023-03-12:
my bad hmmm I should have asked for `log.do_compile`. Can you upload that file instead please

disdi, 2023-03-12:
They are identical

$ diff temp/run.do_compile temp/run.do_compile.2481686
$
$

disdi, 2023-03-12:
Attached -
[log.do_compile.txt](https://github.com/riscv/meta-riscv/files/10950251/log.do_compile.txt)

kraj, 2023-03-14:
I can not reproduce this failure sadly here. Can you post your output at the top of bitbake when it spits out `Build Configuration`

disdi, 2023-03-15:
Pasting screenshot of my bitbake build configuration :
[image]

kraj, 2023-03-15:
OK this seems to be fresh tip of trunk for all the repos. Are you still seeing the nfs-utils build failure with this. ?

kraj, 2023-03-16:
I can not reproduce this at all. I used same layer at same revs and it all built fine. I wonder if it makes a difference if you try a different build host.
```txt
% bitbake nfs-utils
NOTE: Started PRServer with DBfile: /mnt/b/yoe/master/cache/prserv.sqlite3, Address: 127.0.0.1:41655, PID: 1574013
Loading cache: 100% |##################################################################################################################################################################################| Time: 0:00:00
Loaded 4269 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.3.1"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv64-yoe-linux"
MACHINE              = "qemuriscv64"
DISTRO               = "yoe"
DISTRO_VERSION       = "2023.02"
TUNE_FEATURES        = "riscv64"
meta-yoe             = "kraj/master:d8f92c84f2716afea7a1e01ddba6565b2814a094"
meta-riscv           = "HEAD:880aa9b7e39502d196dcaf296cbf0efe0d585f50"
meta-elm-binary      = "yoe/mut:48bc57bb3e0e9befa5716ca75158ba7cc27204bf"
meta-multimedia
meta-networking
meta-oe
meta-python          = "HEAD:093077adc12e8b82eefe8feaae2c07224fd68a34"
meta                 = "HEAD:37b5d544d34c9caaf70def0a29dfb0cf896dc1b7"

Initialising tasks: 100% |#############################################################################################################################################################################| Time: 0:00:01
Sstate summary: Wanted 155 Local 154 Mirrors 0 Missed 1 Current 365 (99% match, 99% complete)
NOTE: Executing Tasks
NOTE: Tasks Summary: Attempted 1889 tasks of which 1889 didn't need to be rerun and all succeeded.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 4 seconds
```

disdi, 2023-03-17:
> OK this seems to be fresh tip of trunk for all the repos. Are you still seeing the nfs-utils build failure with this. ?

I still see the same failures.

I am able to build nfs-utils which same risc-v toolchain as a standalone project.

kraj, 2023-03-17:
is it possible for you to build on a different machine/OS perhaps later version of ubuntu

closed 2023-05-15 by disdi