riscv/meta-riscv #408: BrokenPipeError: [Errno 32] Broken pipe

devPeete, 2023-05-02:
**Description**
I get the following errors when running bitbake (invoked by MACHINE=qemuriscv64 bitbake core-image-full-cmdline):

/usr/lib/python3/dist-packages/html5lib/_trie/_base.py:3: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated since Python 3.3, and in 3.10 it will stop working
  from collections import Mapping
Loading cache: 100% |###############################################################################################################################################################################| Time: 0:00:00
Loaded 1797 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.4.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv64-oe-linux"
MACHINE              = "qemuriscv64"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv64"
meta                 = "work:2ee144a0bfb88823bfa788697bb7afc9a572c413"

ERROR: An uncaught exception occurred in runqueue########################################################################                                                                           | ETA:  0:00:00
Traceback (most recent call last):
  File "/home/leipep/riscv-yocto/openembedded-core/bitbake/lib/bb/runqueue.py", line 1600, in RunQueue.execute_runqueue():
             try:
    >            return self._execute_runqueue()
             except bb.runqueue.TaskFailure:
  File "/home/leipep/riscv-yocto/openembedded-core/bitbake/lib/bb/runqueue.py", line 1550, in RunQueue._execute_runqueue():
                 self.rqdata.init_progress_reporter.next_stage()
    >            self.start_worker()
                 self.rqdata.init_progress_reporter.next_stage()
  File "/home/leipep/riscv-yocto/openembedded-core/bitbake/lib/bb/runqueue.py", line 1378, in RunQueue.start_worker():
             for mc in self.rqdata.dataCaches:
    >            self.worker[mc] = self._start_worker(mc)
     
  File "/home/leipep/riscv-yocto/openembedded-core/bitbake/lib/bb/runqueue.py", line 1351, in RunQueue._start_worker(mc='', fakeroot=False, rqexec=None):
             worker.stdin.write(b"<extraconfigdata>" + pickle.dumps(self.cooker.extraconfigdata) + b"</extraconfigdata>")
    >        worker.stdin.write(b"<workerdata>" + pickle.dumps(workerdata) + b"</workerdata>")
             worker.stdin.flush()
BrokenPipeError: [Errno 32] Broken pipe

ERROR: Running idle function
Traceback (most recent call last):
  File "/home/leipep/riscv-yocto/openembedded-core/bitbake/lib/bb/cooker.py", line 1559, in buildTargetsIdle(server=<bb.server.process.ProcessServer object at 0x7fd7a719c130>, rq=<bb.runqueue.RunQueue object at 0x7fd79d901ca0>, halt=False):
                 try:
    >                retval = rq.execute_runqueue()
                 except runqueue.TaskFailure as exc:
  File "/home/leipep/riscv-yocto/openembedded-core/bitbake/lib/bb/runqueue.py", line 1600, in RunQueue.execute_runqueue():
             try:
    >            return self._execute_runqueue()
             except bb.runqueue.TaskFailure:
  File "/home/leipep/riscv-yocto/openembedded-core/bitbake/lib/bb/runqueue.py", line 1550, in RunQueue._execute_runqueue():
                 self.rqdata.init_progress_reporter.next_stage()
    >            self.start_worker()
                 self.rqdata.init_progress_reporter.next_stage()
  File "/home/leipep/riscv-yocto/openembedded-core/bitbake/lib/bb/runqueue.py", line 1378, in RunQueue.start_worker():
             for mc in self.rqdata.dataCaches:
    >            self.worker[mc] = self._start_worker(mc)
     
  File "/home/leipep/riscv-yocto/openembedded-core/bitbake/lib/bb/runqueue.py", line 1351, in RunQueue._start_worker(mc='', fakeroot=False, rqexec=None):
             worker.stdin.write(b"<extraconfigdata>" + pickle.dumps(self.cooker.extraconfigdata) + b"</extraconfigdata>")
    >        worker.stdin.write(b"<workerdata>" + pickle.dumps(workerdata) + b"</workerdata>")
             worker.stdin.flush()
BrokenPipeError: [Errno 32] Broken pipe

Summary: There were 2 ERROR messages, returning a non-zero exit code.

**Steps to reproduce the issue:**
1. RUN: MACHINE=qemuriscv64 bitbake core-image-full-cmdline

**Describe the results you received:**
Error Message

**Describe the results you expected:**
RISCV Yocto for QEmu

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**
Ubuntu 18.04, GCC 9, Python3.8

kraj, 2023-05-02:
Please use newer version of ubuntu, it may not work well with ubuntu 18.04 python. Or you can also install buildtools-tarball to get newer python version

devPeete, 2023-05-25:
This helped, Thank you.
One mor equestion. Do you know which UART Address is used and where I could possibly change it? I am trying to port this for another simulator.

closed 2024-06-14 by kraj