riscv/meta-riscv #415: weston does not start on Visionfive2

kraj, 2023-06-21:
While X11 images e.g. ( core-image-sato ) boot into graphical interface, `core-image-weston` does not and it ends up with following error. It seems mesa-pvr which provides libEGL.so and binary blob  (visionfive2-pvr-graphics) to provide `libGLESv2.so` and weston can not interact with binary blob from visionfive2-pvr-graphics

I have seen it being reported to work ok on archlinux http://forum.rvspace.org/t/arch-linux-image-for-visionfive-2/1459/175
@cordlandwehr any ideas ?

```
Jun 21 03:53:16 visionfive2 weston[280]: Date: 2023-06-21 UTC
Jun 21 03:53:16 visionfive2 weston[280]: [03:53:15.705] warning: failed to query rendering device from EGL
Jun 21 03:53:16 visionfive2 weston[280]: [03:53:15.705] EGL version: 1.4
Jun 21 03:53:16 visionfive2 weston[280]: [03:53:15.705] EGL vendor: Mesa Project
Jun 21 03:53:16 visionfive2 weston[280]: [03:53:15.705] EGL client APIs: OpenGL OpenGL_ES
Jun 21 03:53:16 visionfive2 weston[280]: [03:53:15.705] warning: Disabling render GPU timeline and explicit synchronization due to missing EGL_ANDROID_native_fence_sync extension
Jun 21 03:53:16 visionfive2 weston[280]: [03:53:15.705] EGL features:
Jun 21 03:53:16 visionfive2 weston[280]:                EGL Wayland extension: yes
Jun 21 03:53:16 visionfive2 weston[280]:                context priority: no
Jun 21 03:53:16 visionfive2 weston[280]:                buffer age: yes
Jun 21 03:53:16 visionfive2 weston[280]:                partial update: no
Jun 21 03:53:16 visionfive2 weston[280]:                swap buffers with damage: no
Jun 21 03:53:16 visionfive2 weston[280]:                configless context: yes
Jun 21 03:53:16 visionfive2 weston[280]:                surfaceless context: yes
Jun 21 03:53:16 visionfive2 weston[280]:                dmabuf support: modifiers
Jun 21 03:53:16 visionfive2 weston[280]: [03:53:16.021] warning: failed to detect GLES version, defaulting to 2.0.
Jun 21 03:53:16 visionfive2 weston[280]: [03:53:16.021] GL version: (null)
Jun 21 03:53:16 visionfive2 weston[280]: [03:53:16.021] GLSL version: (null)
Jun 21 03:53:16 visionfive2 weston[280]: [03:53:16.021] GL vendor: (null)
Jun 21 03:53:16 visionfive2 weston[280]: [03:53:16.021] GL renderer: (null)
Jun 21 03:53:16 visionfive2 weston[280]: [03:53:16.021] Retrieving GL extension string failed.
Jun 21 03:53:16 visionfive2 weston[280]: [03:53:16.021] EGL error state: EGL_SUCCESS (0x3000)
Jun 21 03:53:16 visionfive2 weston[280]: [03:53:16.022] failed to initialize egl
Jun 21 03:53:16 visionfive2 weston[280]: [03:53:16.023] fatal: failed to create compositor backend
Jun 21 03:53:16 visionfive2 weston[280]: Internal warning: debug scope 'drm-backend' has not been destroyed.
```

cordlandwehr, 2023-06-21:
@kraj could you maybe recheck with https://github.com/riscv/meta-riscv/pull/416 and also have a look if you see any warnings/errors during the very early Kernel loading stage when the GPU driver is initialized? I saw similar effects (not really the same, but I did not test against Weston though) when the GPU driver was not initialized correctly due to the named race condition in binary blob loading.

kraj, 2023-06-21:
I had locally made the change to get new pvr driver but kernel update was not there. I have staged your pull locally and see how it goes.

kraj, 2023-06-21:
https://github.com/riscv/meta-riscv/pull/416 helps in bringing up weston.

closed 2023-06-21 by kraj