riscv/meta-riscv #425: glibc build failed on ARCH riscv32

attina, 2023-08-11:
glibc build failed on ARCH riscv32, the the build is ok with same code just change the target ARCH riscv64.

**bitbake configure:**
```
Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "debian-11"
TARGET_SYS           = "riscv32-poky-linux"
MACHINE              = "pc805"
DISTRO               = "poky"
DISTRO_VERSION       = "3.1.23"
TUNE_FEATURES        = "riscv32"
meta-networking      
meta-oe              
meta-python          = "HEAD:e707e9b7cf5c62bff4fee029965a87b22dd4ccba"
meta-pico            = "<unknown>:<unknown>"
meta-riscv           = "HEAD:6f039004c2fa2be041c73d6ff711c86ae139faca"
meta                 
meta-poky            
meta-yocto-bsp       = "HEAD:aec83663aadc871354b441eef0a3a41eb3780b13"
```

**build output:**
```
DEBUG: Executing shell function autotools_preconfigure
DEBUG: Shell function autotools_preconfigure finished
DEBUG: Executing python function autotools_aclocals
DEBUG: SITE files ['endian-little', 'bit-32', 'riscv-common', 'common-linux', 'common-glibc', 'riscv32-linux', 'riscv32-linux', 'common']
DEBUG: Python function autotools_aclocals finished
DEBUG: Executing shell function do_configure
NOTE: Running ../git/configure  --build=x86_64-linux 		  --host=riscv32-poky-linux 		  --target=riscv32-poky-linux 		  --prefix=/usr 		  --exec_prefix=/usr 		  --bindir=/usr/bin 		  --sbindir=/usr/sbin 		  --libexecdir=/usr/libexec 		  --datadir=/usr/share 		  --sysconfdir=/etc 		  --sharedstatedir=/com 		  --localstatedir=/var 		  --libdir=/usr/lib 		  --includedir=/usr/include 		  --oldincludedir=/usr/include 		  --infodir=/usr/share/info 		  --mandir=/usr/share/man 		  --disable-silent-rules 		  --disable-dependency-tracking 		  --with-libtool-sysroot=/build/tmp/work/riscv32-poky-linux/glibc/2.31+gitAUTOINC+d4b7559457-r0/recipe-sysroot --enable-kernel=5.4                 --disable-profile                 --disable-debug --without-gd                 --enable-clocale=gnu                 --with-headers=/build/tmp/work/riscv32-poky-linux/glibc/2.31+gitAUTOINC+d4b7559457-r0/recipe-sysroot/usr/include                 --without-selinux                 --enable-tunables                 --enable-bind-now                 --enable-stack-protector=strong                 --enable-stackguard-randomization                 --disable-crypt                 --with-default-link                 --enable-nscd                                                     --disable-static  
checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv32-poky-linux-gnu
checking for riscv32-poky-linux-gcc... riscv32-poky-linux-gcc     --sysroot=/build/tmp/work/riscv32-poky-linux/glibc/2.31+gitAUTOINC+d4b7559457-r0/recipe-sysroot
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether riscv32-poky-linux-gcc     --sysroot=/build/tmp/work/riscv32-poky-linux/glibc/2.31+gitAUTOINC+d4b7559457-r0/recipe-sysroot accepts -g... yes
checking for gcc... gcc 
checking for riscv32-poky-linux-readelf... riscv32-poky-linux-readelf
checking whether we are using the GNU C++ compiler... yes
checking whether riscv32-poky-linux-g++     --sysroot=/build/tmp/work/riscv32-poky-linux/glibc/2.31+gitAUTOINC+d4b7559457-r0/recipe-sysroot accepts -g... yes
checking whether riscv32-poky-linux-g++     --sysroot=/build/tmp/work/riscv32-poky-linux/glibc/2.31+gitAUTOINC+d4b7559457-r0/recipe-sysroot can link programs... no
checking for sysdeps preconfigure fragments... aarch64 alpha arm csky hppa i386 m68k microblaze checking for grep that handles long lines and -e... /build/tmp/hosttools/grep
checking for egrep... /build/tmp/hosttools/grep -E
mips nios2 powerpc riscv glibc does not yet support 32-bit systems
NOTE: The following config.log files may provide further information.
NOTE: /build/tmp/work/riscv32-poky-linux/glibc/2.31+gitAUTOINC+d4b7559457-r0/build-riscv32-poky-linux/config.log
ERROR: configure failed
WARNING: exit code 1 from a shell command.
ERROR: Execution of '/build/tmp/work/riscv32-poky-linux/glibc/2.31+gitAUTOINC+d4b7559457-r0/temp/run.do_configure.427453' failed with exit code 1
```
**here is the config log file.**
[config.log](https://github.com/riscv/meta-riscv/files/12319898/config.log)

alistair23, 2023-08-11:
It looks like you are using glibc 2.31, which I'm pretty sure is too old to support 32-bit RISC-V.

You should update your `meta` layer to get a later glibc version

kraj, 2023-08-11:
IIRC rv32 port was added in 2.33 release onwards. The build seems to be using `dunfell` release. I would recommend to use `kirkstone` if you are needing LTS, or ideally latest release which is `mickledore`

attina, 2023-08-15:
Thank you very much. @alistair23 @kraj 
Updated to kirkstone solved this problem.

closed 2023-08-15 by attina