riscv/meta-riscv #434: build failing on start with 'ERROR: Nothing RPROVIDES 'systemd''

pfarwsi, 2023-09-25:
**Description**

**Steps to reproduce the issue:**
1. mkdir riscv-yocto && cd riscv-yocto
2. repo init -u https://github.com/riscv/meta-riscv  -b master -m tools/manifests/riscv-yocto.xml
3. repo sync
4. repo start work --all
5. . ./meta-riscv/setup.sh
6. MACHINE=beaglev-starlight-jh7100 bitbake core-image-minimal

**Describe the results you received:**
Loading cache: 100% |                                                                                                                                                        | ETA:  --:--:--
Loaded 0 entries from dependency cache.
Parsing recipes: 100% |#######################################################################################################################################################| Time: 0:01:20
Parsing of 2729 .bb files complete (0 cached, 2729 parsed). 4473 targets, 191 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing RPROVIDES 'systemd' (but /home/pratik/riscv-yocto/openembedded-core/meta/recipes-core/packagegroups/packagegroup-core-boot.bb RDEPENDS on or otherwise requires it)
systemd was skipped: missing required distro feature 'usrmerge' (not in DISTRO_FEATURES)
NOTE: Runtime target 'systemd' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['systemd']
NOTE: Runtime target 'packagegroup-core-boot' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-core-boot', 'systemd']
ERROR: Required build target 'core-image-minimal' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-minimal', 'packagegroup-core-boot', 'systemd']

Summary: There were 2 ERROR messages, returning a non-zero exit code.

**Describe the results you expected:**

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

kraj, 2023-09-25:
Please test the patch proposed in - https://github.com/riscv/meta-riscv/pull/435

closed 2023-09-25 by kraj

pfarwsi, 2023-09-26:
The patch works and build goes ahead. I am testing the build with VisionFive2 as i have that Machine with me. I will test it more and let you know of more changes.