riscv/meta-riscv #462: No hdmi on mangopi

skypap, 2024-02-29:
**Description**

**Steps to reproduce the issue:**
1.Bake mango pi with default param
2. The board had no interraction with de hdmi port

**Describe the results you expected:**
I mean there should be some message from dmesg like 
[sun20i_d1_hdmi_phy_enable]:tx_ready_status
etc...

am I missing something???

kraj, 2024-02-29:
are you able to see login prompt on serial console ?

skypap, 2024-03-05:
Sure

U-Boot SPL 2024.01-rc1 (Oct 31 2023 - 08:53:02 +0000)
mxstatus=0xc0408000 mhcr=0x0000010b m�
OpenSBI v1.4
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name             : MangoPi MQ Pro
Platform Features         : medeleg
Platform HART Count       : 1                                                   
Platform IPI Device       : ---                                                 
Platform Timer Device     : --- @ 0Hz                                           
Platform Console Device   : uart8250                                            
Platform HSM Device       : sun20i-d1-ppu                                       
Platform PMU Device       : thead,c900-pmu                                      
Platform Reboot Device    : sunxi-wdt-reset                                     
Platform Shutdown Device  : ---                                                 
Platform Suspend Device   : ---                                                 
Platform CPPC Device      : ---                                                 
Firmware Base             : 0x40000000                                          
Firmware Size             : 323 KB                                              
Firmware RW Offset        : 0x40000                                             
Firmware RW Size          : 67 KB                                               
Firmware Heap Offset      : 0x48000                                             
Firmware Heap Size        : 35 KB (total), 2 KB (reserved), 9 KB (used), 24 KB )
Firmware Scratch Size     : 4096 B (total), 312 B (used), 3784 B (free)         
Runtime SBI Version       : 2.0                                                 
                                                                                
Domain0 Name              : root                                                
Domain0 Boot HART         : 0                                                   
Domain0 HARTs             : 0*                                                  
Domain0 Region00          : 0x0000000002500000-0x0000000002500fff M: (I,R,W) S/)
Domain0 Region01          : 0x0000000040040000-0x000000004005ffff M: (R,W) S/U:)
Domain0 Region02          : 0x0000000040000000-0x000000004003ffff M: (R,X) S/U:)
Domain0 Region03          : 0x0000000010000000-0x0000000013ffffff M: (I,R,W) S/)
Domain0 Region04          : 0x0000000000000000-0xffffffffffffffff M: () S/U: (R)
Domain0 Next Address      : 0x0000000042e00000                                  
Domain0 Next Arg1         : 0x0000000042e8d070                                  
Domain0 Next Mode         : S-mode                                              
Domain0 SysReset          : yes                                                 
Domain0 SysSuspend        : yes                                                 
                                                                                
Boot HART ID              : 0                                                   
Boot HART Domain          : root                                                
Boot HART Priv Version    : v1.11                                               
Boot HART Base ISA        : rv64imafdcvx                                        
Boot HART ISA Extensions  : zicntr,zihpm                                        
Boot HART PMP Count       : 8                                                   
Boot HART PMP Granularity : 11 bits                                             
Boot HART PMP Address Bits: 38                                                  
Boot HART MHPM Info       : 12 (0x0003e3f8)                                     
Boot HART MIDELEG         : 0x0000000000020222                                  
Boot HART MEDELEG         : 0x000000000000b109                                  
                                                                                
                                                                                
U-Boot 2024.01-rc1 (Oct 31 2023 - 08:53:02 +0000) Allwinner Technology          
                                                                                
DRAM:  1 GiB                                                                    
Core:  48 devices, 20 uclasses, devicetree: separate                            
WDT:   Started watchdog@6011000 with servicing every 1000ms (16s timeout)       
MMC:   mmc@4020000: 0, mmc@4021000: 1                                           
Loading Environment from FAT... Unable to read "uboot.env" from mmc0:1...       
In:    serial@2500000                                                           
[... 476 lines cut ...]
[  OK  ] Reached target Bluetooth Support.                                      
         Starting Hostname Service...                                           
[  OK  ] Started User Login Management.                                         
[  OK  ] Started Network Configuration.                                         
[  OK  ] Reached target Network.                                                
         Starting Permit User Sessions...                                       
[  OK  ] Finished Permit User Sessions.                                         
[... cut ...]
                                                                                
Poky (Yocto Project Reference Dis                                               
Poky (Yocto Projtro) 4.3+snapshot-25d60ac6f61cc186b4c5a961554bee583736fd17 mang0
                                                                                
mangopi-mq-pro login: ect Reference Distro) 4.3+snapshot-25d60ac6f61cc186b4c5a90
                                                                                
mangopi-mq-pro login: 

I hope I didnt do anything stupid :S

kraj, 2024-06-14:
perhaps look into device tree and check if hdmi is enabled or not.

kraj, 2026-10-12:
@michaelopdenacker are you able to comment if this issue is still open ?

michaelopdenacker, 2026-10-13:
We don't support the MangoPi board any more.
I think we should close this bug.

kraj, 2026-10-13:
Board Not supported anymore

closed 2026-10-13 by kraj