riscv/meta-riscv #465: [master: visionfive2] mesa-pvr

jwinarske, 2024-03-20:
**Description**

Running `vulkaninfo` on `core-image-weston` (x11 removed) none of the extensions required to present to display are present:
```
VK_KHR_display
VK_KHR_wayland_surface
VK_KHR_xcb_surface
VK_KHR_xlib_surface
```

**Steps to reproduce the issue:**
```
MACHINE ?= "visionfive2"

INIT_MANAGER = "systemd"
DISTRO_FEATURES:append = " systemd alsa usbhost pipewire"
DISTRO_FEATURES:remove = "sysvinit usbgadget ptest xen x11"
DISTRO_FEATURES:append = " opengl wayland pam vulkan"

IMAGE_INSTALL:append = " \
    \
    curl \
    vulkan-tools \
    vulkan-caps-viewer \
    sascha-samples \
    vkcube \
    visionfive2-pvr-graphics \
    \
    libsf-codaj12 \
    libsf-omxil \
    libsf-wave420l \
    libsf-wave511 \
    jpu-module \
    vdec-module \
    venc-module \
"

PACKAGECONFIG:append:pn-mesa-pvr = " egl"
PACKAGECONFIG:remove:pn-qtbase = " gl"
PACKAGECONFIG:append:pn-qtbase = " gles2"
```

**Describe the results you received:**

**Describe the results you expected:**

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

https://github.com/jwinarske/meta-vulkan

jwinarske, 2024-03-20:
Looks like upstream has Vulkan still as a work in progress:
https://rvspace.org/en/project/VisionFive2_Debian_Wiki_202403_Release

jwinarske, 2024-03-23:
Upstream working reference has Vulkan display/surface support, as well as working libcamera IPU (lagging libcamera 0.2.0).  They are moving to 6.6 kernel soon with most all of their kernel drivers upstreamed:
https://github.com/starfive-tech/VisionFive2

The Pine64 STAR64 and PineTab-V use the same SoC and are based on the starfive-tech VisionFive2 repo:
https://pine64.org/devices/star64
https://pine64.org/devices/pinetab-v

If you're open to PRs I'm happy to make some updates.

kraj, 2024-03-23:
@jwinarske yes, I am looking for updating to latest stable kernel as the upcoming yocto 5 release is a LTS one, and it will be good to have it on latest LTS of every package we can

jwinarske, 2024-03-23:
Upstream is using `MESA3D_VERSION = 20.2.2`:
https://github.com/starfive-tech/buildroot/blob/7dd07bb55ed4f4f1507b4f2d0b1e95d39d94c0f2/package/mesa3d/mesa3d.mk

Their libcamera is behind 0.2.0 as well.

I'll take a look at re-basing mesa + libcamera to scarthgap versions.

kraj, 2026-10-12:
@jwinarske I wonder if you have updates on this issue ?

jwinarske, 2026-10-12:
@kraj It would need a build and re-check.  Let me do that, and update this issue.

jwinarske, 2026-10-13:
@kraj 

Re-checked against master (meta-riscv `0a95d9e`, `MACHINE=visionfive2`): inspected the PVR blob master pins, and ran `vulkaninfo` on a VF2 with the StarFive vendor image.

**Root cause: vendor Vulkan ICD has no WSI.** Not fixable in `mesa-pvr`.

- ICD: `libVK_IMG.so.1.19.6345021` from `soft_3rdpart` @ `b60da16` (`IMG_GPU/out/img-gpu-powervr-bin-1.19.6345021.tar.gz`). Last `IMG_GPU` update upstream is `1612caa` (2023-10-18); we're 60 commits past it, so this is the newest blob.
- Extension strings in the ICD: ~90 `VK_KHR_*`/`VK_EXT_*`, none of `VK_KHR_surface`, `VK_KHR_swapchain`, `VK_KHR_display`, `VK_KHR_{wayland,xcb,xlib}_surface`.
- No reference to `libpvr_mesa_wsi.so` / `pvr_mesa_wsi_sym_addr`; no `dlopen`/`dlsym` imports. DT_NEEDED: `libsrv_um libusc libufwriter libm libdrm libc`.
- `mesa-pvr` 0049 (`pvr/wsi`) builds `libpvr_mesa_wsi.so` only with `-Dvulkan-drivers=pvr`; we pass `''`. Enabling it gives a lib nothing loads.
- meta-starfive (`starfive-scarthgap`) `mesa-pvr.inc` does the same (`VULKAN_DRIVERS = ""`), so vendor Yocto images have the same gap.

```
$ strings -n4 libVK_IMG.so.1.19.6345021 | grep -E '^VK_KHR_(surface|swapchain|display|wayland_surface|xcb_surface|xlib_surface)$'
$ strings -n4 libVK_IMG.so.1.19.6345021 | grep -i pvr_mesa_wsi
$ nm -D --undefined-only libVK_IMG.so.1.19.6345021 | grep -E 'dlopen|dlsym'
$
```

On target (VF2, StarFive Ubuntu 24.04.3, `6.12.5-starfive`, `img-pvr-rogue 1.19-1`; same DDK `1.19@6345021`, different build than the Yocto blob, same result):

```
$ VK_DRIVER_FILES=/etc/vulkan/icd.d/icdconf.json vulkaninfo --summary
Instance Extensions: count = 9
VK_EXT_debug_report
VK_EXT_debug_utils
VK_KHR_device_group_creation
VK_KHR_external_fence_capabilities
VK_KHR_external_memory_capabilities
VK_KHR_external_semaphore_capabilities
VK_KHR_get_physical_device_properties2
VK_KHR_portability_enumeration
VK_LUNARG_direct_driver_loading
...
deviceName = PowerVR B-Series BXE-4-32
driverInfo = 1.19@6345021
$ vulkaninfo | sed -n '/Device Extensions/,/^$/p' | grep -E 'count|swapchain'
Device Extensions: count = 86
$
```

Note: without `VK_DRIVER_FILES`, `vulkaninfo` on the vendor Ubuntu image lists `VK_KHR_surface`/`VK_KHR_display`/`VK_KHR_wayland_surface` etc. at instance level. Those come from Mesa's `libvulkan_radeon.so` ICD (`mesa-vulkan-drivers` 22.3.6) also being installed, not the PVR ICD. The PVR device still has no `VK_KHR_swapchain`.

Headless Vulkan (compute/offscreen) is what this blob supports. Display/surface needs one of:

1. StarFive/IMG ship a DDK build with WSI enabled (ICD that loads `libpvr_mesa_wsi.so`). Needs a request to StarFive.
2. Open stack: mainline `drm/imagination` + Mesa `imagination` Vulkan. JH7110 GPU is BXE-4-32 `36.50.54.182`, listed by Mesa as unsupported/not under active development; firmware exists. Scoping this as an experiment on `visionfive2` (same pattern as `licheepi-4a`: `VULKAN_DRIVERS:append:pn-mesa = ",imagination"`, `powervrconf`, `PVR_I_WANT_A_BROKEN_VULKAN_DRIVER=1`). Will report back.

Proposal: keep #465 open as vendor-blocked; no `mesa-pvr` change.

OOHehir, 2026-10-23:
I have opened #737, which may resolve the issue you are seeing @jwinarske. It keeps the vendor blob & adds Arm's [Vulkan WSI layer](https://gitlab.freedesktop.org/mesa/vulkan-wsi-layer), an implicit layer that adds surface & swapchain support on top of an ICD that has none.

On a VisionFive 2 with meta-riscv master, `vkcube` now renders on HDMI under Weston & straight to the display, as a normal user, on sysvinit & on systemd with `pam` as in your setup. It does not cover X11 (`VK_KHR_xcb_surface`, `VK_KHR_xlib_surface`). Results & what was not tested are in the PR.

It is subject to your & the maintainers' feedback, & should complement your open-stack work rather than compete with it: if the Mesa `imagination` driver gains BXE-4-32 support, that is the better long-term answer.