riscv/meta-riscv #469: How can I increase my RAM memory for milk-v duo 64MB?

Markus250, 2024-04-16:
I have Milk-V Duo 64MB, in Linux I see only ~22MB (free and even less).
I find posts like [this](https://community.milkv.io/t/enable-and-add-swap-partition-to-duo-buildroot/411) and [this](https://community.milkv.io/t/menuconfig-milkv-duo-linux-kernel/1112), , but that's all for buildroot (i am using yocto).
How can I use your layer to increase the memory for the Linux kernel to the maximum (well, or somehow divide it differently, in one project I use the RTOS kernel, in another only the camera, and in others I do not use either the RTOS or the camera, but only the network interface, I need all the memory).
I hope I expressed my thoughts correctly, thanks in advance!

kraj, 2024-06-14:
@Markus250 I think there is an external logic in the BR sdk to generate some header file (cvi_board_memmap.h) and conf file (cvi_board_memmap.conf) and linker script (cvi_board_memmap.ld) via a python script `memmap.py`, I think this step s completely missing here in yocto setup. So it needs to be done for yocto. Perhaps they can be directly patched into u-boot/kernel/freestos builds, look into the logic in `memmap.py` perhaps as a starting point

Markus250, 2024-08-28:
I implemented it like this, I wrote my script in python3 which creates file cvi_board_memmap.h, and i just create patch for linux kernel! In my-meta/recipes-kernel/linux/linux-milkv-duo-dev.bbappend i put my patch "file://0001-memmap.patch", and it works!
I’m sharing the script, you never know who will need it:

```python
SIZE_1M = 0x100000
SIZE_1K = 1024

DRAM_BASE = 0x80000000
DRAM_SIZE = 64 * SIZE_1M

# ==============
# C906L FreeRTOS
# ==============
FREERTOS_SIZE = 0 * SIZE_1K
# FreeRTOS is at the end of DRAM
FREERTOS_ADDR = DRAM_BASE + DRAM_SIZE - FREERTOS_SIZE
FSBL_C906L_START_ADDR = FREERTOS_ADDR

# ==============================
# OpenSBI | arm-trusted-firmware
# ==============================
# Monitor is at the begining of DRAM
MONITOR_ADDR = DRAM_BASE

ATF_SIZE = 512 * SIZE_1K
OPENSBI_SIZE = 512 * SIZE_1K
OPENSBI_FDT_ADDR = MONITOR_ADDR + OPENSBI_SIZE

# =========================
# memory@DRAM_BASE in .dts.
# =========================
# Ignore the area of FreeRTOS in u-boot and kernel
KERNEL_MEMORY_ADDR = DRAM_BASE
KERNEL_MEMORY_SIZE = DRAM_SIZE - FREERTOS_SIZE

# =================
# Multimedia buffer. Used by u-boot/kernel/FreeRTOS
# =================
ION_SIZE = 0 * SIZE_1M
H26X_BITSTREAM_SIZE = 0 * SIZE_1M
H26X_ENC_BUFF_SIZE = 0
ISP_MEM_BASE_SIZE = 0 * SIZE_1M
BOOTLOGO_SIZE = 0 * SIZE_1M
FREERTOS_RESERVED_ION_SIZE = H26X_BITSTREAM_SIZE + H26X_ENC_BUFF_SIZE + ISP_MEM_BASE_SIZE + BOOTLOGO_SIZE

# ION after FreeRTOS
ION_ADDR = FREERTOS_ADDR - ION_SIZE

# Buffers of the fast image are inside the ION buffer
H26X_BITSTREAM_ADDR = ION_ADDR
H26X_ENC_BUFF_ADDR = H26X_BITSTREAM_ADDR + H26X_BITSTREAM_SIZE
ISP_MEM_BASE_ADDR = H26X_ENC_BUFF_ADDR + H26X_ENC_BUFF_SIZE

# Boot logo is after ISP buffer and inside the ION buffer
BOOTLOGO_ADDR = ISP_MEM_BASE_ADDR + ISP_MEM_BASE_SIZE

assert BOOTLOGO_ADDR + BOOTLOGO_SIZE <= ION_ADDR + ION_SIZE

# ===================
# FSBL and u-boot-2021
# ===================
CVI_UPDATE_HEADER_SIZE = SIZE_1K
UIMAG_SIZE = 15 * SIZE_1M

# kernel image loading buffer
UIMAG_ADDR = DRAM_BASE + 20 * SIZE_1M
CVI_UPDATE_HEADER_ADDR = UIMAG_ADDR - CVI_UPDATE_HEADER_SIZE

# FSBL decompress buffer
FSBL_UNZIP_ADDR = UIMAG_ADDR
FSBL_UNZIP_SIZE = UIMAG_SIZE

assert UIMAG_ADDR + UIMAG_SIZE <= ION_ADDR

# u-boot's run address and entry point
CONFIG_SYS_TEXT_BASE = DRAM_BASE + 2 * SIZE_1M
# u-boot's init stack point is only used before board_init_f()
CONFIG_SYS_INIT_SP_ADDR = UIMAG_ADDR + UIMAG_SIZE

def to_hex(val):
    value = hex(int(val))
    return value

with open("cvi_board_memmap.h", "w") as f:
    f.write("#ifndef __BOARD_MMAP__b126ca63__\n")
    f.write("#define __BOARD_MMAP__b126ca63__\n\n")
    
    f.write(f"#define CONFIG_SYS_TEXT_BASE {to_hex(CONFIG_SYS_TEXT_BASE)}\n")
    f.write(f"#define CVIMMAP_ATF_SIZE {to_hex(ATF_SIZE)}\n")
    f.write(f"#define CVIMMAP_BOOTLOGO_ADDR {to_hex(BOOTLOGO_ADDR)}\n")
    f.write(f"#define CVIMMAP_BOOTLOGO_SIZE {to_hex(BOOTLOGO_SIZE)}\n")
    f.write(f"#define CVIMMAP_CONFIG_SYS_INIT_SP_ADDR {to_hex(CONFIG_SYS_INIT_SP_ADDR)}\n")
    f.write(f"#define CVIMMAP_CVI_UPDATE_HEADER_ADDR {to_hex(CVI_UPDATE_HEADER_ADDR)}\n")
    f.write(f"#define CVIMMAP_CVI_UPDATE_HEADER_SIZE {to_hex(CVI_UPDATE_HEADER_SIZE)}\n")
    f.write(f"#define CVIMMAP_DRAM_BASE {to_hex(DRAM_BASE)}\n")
    f.write(f"#define CVIMMAP_DRAM_SIZE {to_hex(DRAM_SIZE)}\n")
    f.write(f"#define CVIMMAP_FREERTOS_ADDR {to_hex(FREERTOS_ADDR)}\n")
    f.write(f"#define CVIMMAP_FREERTOS_RESERVED_ION_SIZE {to_hex(FREERTOS_RESERVED_ION_SIZE)}\n")
    f.write(f"#define CVIMMAP_FREERTOS_SIZE {to_hex(FREERTOS_SIZE)}\n")
    f.write(f"#define CVIMMAP_FSBL_C906L_START_ADDR {to_hex(FSBL_C906L_START_ADDR)}\n")
    f.write(f"#define CVIMMAP_FSBL_UNZIP_ADDR {to_hex(FSBL_UNZIP_ADDR)}\n")
    f.write(f"#define CVIMMAP_FSBL_UNZIP_SIZE {to_hex(FSBL_UNZIP_SIZE)}\n")
    f.write(f"#define CVIMMAP_H26X_BITSTREAM_ADDR {to_hex(H26X_BITSTREAM_ADDR)}\n")
    f.write(f"#define CVIMMAP_H26X_BITSTREAM_SIZE {to_hex(H26X_BITSTREAM_SIZE)}\n")
    f.write(f"#define CVIMMAP_H26X_ENC_BUFF_ADDR {to_hex(H26X_ENC_BUFF_ADDR)}\n")
    f.write(f"#define CVIMMAP_H26X_ENC_BUFF_SIZE {to_hex(H26X_ENC_BUFF_SIZE)}\n")
    f.write(f"#define CVIMMAP_ION_ADDR {to_hex(ION_ADDR)}\n")
    f.write(f"#define CVIMMAP_ION_SIZE {to_hex(ION_SIZE)}\n")
    f.write(f"#define CVIMMAP_ISP_MEM_BASE_ADDR {to_hex(ISP_MEM_BASE_ADDR)}\n")
    f.write(f"#define CVIMMAP_ISP_MEM_BASE_SIZE {to_hex(ISP_MEM_BASE_SIZE)}\n")
    f.write(f"#define CVIMMAP_KERNEL_MEMORY_ADDR {to_hex(KERNEL_MEMORY_ADDR)}\n")
    f.write(f"#define CVIMMAP_KERNEL_MEMORY_SIZE {to_hex(KERNEL_MEMORY_SIZE)}\n")
    f.write(f"#define CVIMMAP_MONITOR_ADDR {to_hex(MONITOR_ADDR)}\n")
    f.write(f"#define CVIMMAP_OPENSBI_FDT_ADDR {to_hex(OPENSBI_FDT_ADDR)}\n")
    f.write(f"#define CVIMMAP_OPENSBI_SIZE {to_hex(OPENSBI_SIZE)}\n")
    f.write(f"#define CVIMMAP_UIMAG_ADDR {to_hex(UIMAG_ADDR)}\n")
    f.write(f"#define CVIMMAP_UIMAG_SIZE {to_hex(UIMAG_SIZE)}\n\n")

    f.write("#endif /* __BOARD_MMAP__b126ca63__ */\n")

closed 2024-08-28 by Markus250