riscv/meta-riscv #478: Stuck at "NOTE: Starting bitbake server..."

VEGA-Processor, 2024-06-14:
**Description**

**Steps to reproduce the issue:**

1. mkdir riscv-yocto && cd riscv-yocto
repo init -u https://github.com/riscv/meta-riscv  -b master -m tools/manifests/riscv-yocto.xml
repo sync
repo start work --all

2. repo sync
repo rebase

3. . ./meta-riscv/setup.sh

**Describe the results you received:**

After running  ". ./meta-riscv/setup.sh" getting below log:

Init OE
This is the default build configuration for the Poky reference distribution.

\### Shell environment set up for builds. \###

You can now run 'bitbake <target>'

Common targets are:
    core-image-minimal
    core-image-full-cmdline
    core-image-sato
    core-image-weston
    meta-toolchain
    meta-ide-support

You can also run generated qemu images with a command like 'runqemu qemux86-64'.

Other commonly useful commands are:
 - 'devtool' and 'recipetool' handle common recipe tasks
 - 'bitbake-layers' handles common layer tasks
 - 'oe-pkgdata-util' handles common target package tasks
Adding layers
NOTE: Starting bitbake server...

**Describe the results you expected:**
Setup Build Environment for Yocto  should have completed

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

kraj, 2024-06-14:
there must be a file called `bitbake-cookerdaemon.log` in the build tree. This should have some more details about why its getting stuck.

VEGA-Processor, 2024-11-12:
[bitbake-cookerdaemon.log](https://github.com/user-attachments/files/17716104/bitbake-cookerdaemon.log)
bitbake-cookerdaemon.log file attached

kraj, 2024-11-19:
it seems to be stuck in bitbake-layers add-layer meta-python command. I wonder if its a python version issue. Can you share details about the OS on your build host and python version installed on it.

VEGA-Processor, 2024-11-20:
### **OS version**
Static hostname: hdg145
Icon name: computer-desktop
Chassis: desktop 🖥️
Machine ID: 31ad43890c074021a576177b8db0aca3
Boot ID: 9771a6c1901540e3a2085983f0a3adec
Operating System: Ubuntu 24.04.1 LTS              
Kernel: Linux 6.8.0-48-generic
Architecture: x86-64
Hardware Vendor: Dell Inc.
Hardware Model: OptiPlex 5090
Firmware Version: 1.29.0
Firmware Date: Mon 2024-09-02
Firmware Age: 2month 2w 4d

**### Python version**
python --version
Command 'python' not found, did you mean:
  command 'python3' from deb python3
  command 'python' from deb python-is-python3

python3 --version
Python 3.12.3

closed 2025-02-13 by VEGA-Processor