riscv/meta-riscv #479: Milkv duo 256M support

lin-calvin, 2024-06-24:
**Description**
Will Milkv duo 256M be supported?

**Steps to reproduce the issue:**
1.
2.
3.

**Describe the results you received:**

**Describe the results you expected:**

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

lazyparser, 2024-06-24:
Hi @lin-calvin 

I'm glad you're interested in Duo 256M. I can help find volunteers to support it if needed. Or you can try to modify it and submit a PR! It's not hard once you get the basic technical details.

lin-calvin, 2024-07-08:
> Hi @lin-calvin
> 
> I'm glad you're interested in Duo 256M. I can help find volunteers to support it if needed. Or you can try to modify it and submit a PR! It's not hard once you get the basic technical details.

I may try to do it, If there are any volunteers can help to do it. I will thanks alot

threexc, 2026-09-02:
Should be able to close this now.

PR #486 (2024-08-27): Add support for Milk-V Duo 256M and Duo S boards
fixes #479
closed 2026-10-12 by kraj