riscv/meta-riscv #487: Milkv duo support NAND image

Markus250, 2024-08-28:
In buildroot I can compile image for SPI NAND next command:
./build.sh milkv-duo-spinand
and in the folder out/
I will receive the files for the firmware nand (boot.spinand, fip.bin, rootfs.spinand, system.spinand)
I load this into nand and it works
How can create an image for nand using this layer (yocto)?

sorry for my eng

ziswiler, 2025-01-31:
This broke the build for me:

```
make: Leaving directory '/var/home/sumo/Sources/riscv-yocto/build/tmp/work/riscv64-poky-linux/milkv-duo-fsbl/1.0/git/firmware'
usage: fiptool.py [-h] [-v] {genfip} ...
fiptool.py: error: unrecognized arguments: --NAND_BOOT=1
WARNING: /var/home/sumo/Sources/riscv-yocto/build/tmp/work/riscv64-poky-linux/milkv-duo-fsbl/1.0/temp/run.do_compile.1911581:150 exit 2 from 'python3 /var/home/sumo/Sources/riscv-yocto/build/tmp/work/riscv64-poky-linux/milkv-duo-fsbl/1.0/git/firmware/plat/cv180x/fiptool.py genfip /var/home/sumo/Sources/riscv-yocto/build/tmp/work/riscv64-poky-linux/milkv-duo-fsbl/1.0/git/firmware/build/fip.bin --MONITOR_RUNADDR=0x80000000 --CHIP_CONF=/var/home/sumo/Sources/riscv-yocto/build/tmp/work/riscv64-poky-linux/milkv-duo-fsbl/1.0/git/firmware/plat/cv180x/chip_conf.bin --NOR_INFO=FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF --NAND_INFO=00000000 --BL2=/var/home/sumo/Sources/riscv-yocto/build/tmp/work/riscv64-poky-linux/milkv-duo-fsbl/1.0/git/firmware/build/cv180x/bl2.bin --BLCP_IMG_RUNADDR=0x05200200 --BLCP_PARAM_LOADADDR=0 --BLCP_2ND=/var/home/sumo/Sources/riscv-yocto/build/tmp/work/riscv64-poky-linux/milkv-duo-fsbl/1.0/git/firmware/build/blank.bin --BLCP_2ND_RUNADDR=0x83f40000 --DDR_PARAM=/var/home/sumo/Sources/riscv-yocto/build/tmp/work/riscv64-poky-linux/milkv-duo-fsbl/1.0/git/firmware/test/cv181x/ddr_param.bin --MONITOR=/var/home/sumo/Sources/riscv-yocto/build/tmp/deploy/images/milkv-duo/fw_dynamic.bin --LOADER_2ND=/var/home/sumo/Sources/riscv-yocto/build/tmp/deploy/images/milkv-duo/u-boot.bin --NAND_BOOT=1'
WARNING: Backtrace (BB generated script): 
        #1: do_compile, /var/home/sumo/Sources/riscv-yocto/build/tmp/work/riscv64-poky-linux/milkv-duo-fsbl/1.0/temp/run.do_compile.1911581, line 150
        #2: main, /var/home/sumo/Sources/riscv-yocto/build/tmp/work/riscv64-poky-linux/milkv-duo-fsbl/1.0/temp/run.do_compile.1911581, line 202

```

Markus250, 2025-02-10:
Yep, It doesn't work, I have exactly the same error
I am check fiptool.py args, he does not have --NAND_BOOT=1
@vishwamartur you make patch forfiptool.py?

kraj, 2025-02-10:
@ziswiler @Markus250 I have removed the non existing option for now. I assume there is a patch for https://github.com/milkv-duo/duo-buildroot-sdk/blob/develop/fsbl/plat/cv181x/fiptool.py which is missing, or the option was added wrongly.

Markus250, 2025-02-11:
@kraj  Yes, without this argument, the assembly of the image has advanced. Also, I had to comment the line 
`cp ${WORKDIR}/fip.bin ${DEPLOYDIR}/` in file `milkv-duo-bootfiles.bb` because in recipe `milkv-duo-bootfiles.bb` and `milkv-duo-fsbl.bb` simultaneous copying of the file `fip.bin`
But it still did not work, files `rootfs.spinand`, `boot.spinand`, etc..  do not appear, do not even gather =(
How much I could make out buildroot this files make python script mkubiimg.py, example from `duo-buildroot-sdk/build/Makefile`:
`${Q}python3 $(COMMON_TOOLS_PATH)/spinand_tool/mkubiimg.py --ubionly $(FLASH_PARTITION_XML) ROOTFS $(OUTPUT_DIR)/rawimages/rootfs.sqsh $(OUTPUT_DIR)/rawimages/rootfs.spinand -b $(CONFIG_NANDFLASH_BLOCKSIZE) -p $(CONFIG_NANDFLASH_PAGESIZE)` 
Again, these are all my assumptions, I could never collect the image of NAND through Yocto 
I will be grateful if you help me)

kraj, 2025-02-13:
@Markus250 does the fix proposed in #511 fix the issue ?

Markus250, 2025-02-18:
@kraj  Well, the assembly is going on, but the image for nand not created =(

kraj, 2025-02-19:
> [@kraj](https://github.com/kraj) Well, the assembly is going on, but the image for nand not created =(

That seems not good. Does it work if we revert the original patch to add NAND support ?

Markus250, 2025-02-26:
> @kraj That seems not good. Does it work if we revert the original patch to add NAND support ?

Unfortunately, no, if you look at how the buildroot does it, then there is more manipulations there, I tried to do it myself, but nothing worked for me

kraj, 2025-02-27:
@Markus250 ok, are you able to build if you revert https://github.com/riscv/meta-riscv/commit/b94502a53caa8efe101a1567eadea8bccdc60198 ?

in meta-riscv you can do

git revert b94502a53caa8efe101a1567eadea8bccdc60198

and build

Markus250, 2025-02-27:
@kraj Yes, but need we must change the following:
`conf/machine/milkv-duo.conf` restore `WKS_FILE ?= "milkv-duo.wks"` instead of `WKS_FILE ?= "milkv-duo-spinand.wks"`
revert old milkv-duo-fsbl.bb from meta-riskv branch scarthgap
it seems like everything, then you can collect an image, but again, this image is not for spinand
in branch scarthgap image build successfully for sd-card, It is very sorry that no one can add the opportunity to collect under spi-nand =(

kraj, 2025-02-27:
if it is breaking then I would like to revert it. Can you post the patch that works for you ?

kraj, 2026-10-12:
does this work now? can we close it?

Markus250, 2026-10-14:
@kraj Unfortunately, I can’t check, a lot of time has passed since then and I ended up doing everything by means of buildroot

AndreyLalaev, 2026-10-14:
@kraj I believe it is still not supported.