riscv/meta-riscv #488: riscv-initramfs-image default password

lyctw, 2024-08-28:
Hi developers,

I've built `riscv-initramfs-image` provided by meta-riscv and added the variables in local.conf

```
INITRAMFS_IMAGE_BUNDLE = "1"
INITRAMFS_IMAGE = "riscv-initramfs-image"
INITRAMFS_IMAGE_NAME = "riscv-initramfs-image-${MACHINE}.rootfs"
BOOT_SPACE = "1073741"
INITRAMFS_MAXSIZE = "315400"
IMAGE_FSTYPES:pn-${INITRAMFS_IMAGE} = "${INITRAMFS_FSTYPES}"
```

it successfully launched login prompt using busybox init. But it asks password for `root`, Am I missing something?

I've also checked `EXTRA_IMAGE_FEATURES = "debug-tweaks"` specified in my local.conf

kraj, 2024-08-29:
Can you try this for the bb file

```
# will be used for the update image
DESCRIPTION = "initramfs updater image"
LICENSE = "MIT"

LIC_FILES_CHKSUM = "file://${COMMON_LICENSE_DIR}/MIT;md5=0835ade698e0bcf8506ecda2f7b4f302"

# Do not pollute the initrd image with rootfs features
IMAGE_FEATURES = ""

# Clear image depends on some machines e.g. riscv opensbi is in EXTRA_IMAGEDEPENDS
# which can cause dependency loops
EXTRA_IMAGEDEPENDS = ""

IMAGE_NAME_SUFFIX = ""

export IMAGE_BASENAME = "riscv-initramfs-image"
IMAGE_LINGUAS = ""

PACKAGE_INSTALL = "\
  base-files \
  base-passwd \
  busybox \
  mtd-utils \
  mtd-utils-ubifs \
  e2fsprogs \
  e2fsprogs-mke2fs \
  e2fsprogs-e2fsck \
  e2fsprogs-tune2fs \
  util-linux-fsck \
  util-linux-fdisk \
  "
IMAGE_FSTYPES = "${INITRAMFS_FSTYPES}"
# Check conf/site.conf where there is IMAGE_FSTYPES:append
# if that value changes make the change here as well
IMAGE_FSTYPES:remove = "wic.* wic.xz wic.bmap tar.xz"
inherit core-image

IMAGE_DEVICE_TABLES = "files/device_table-minimal.txt"
IMAGE_ROOTFS_SIZE = "8192"
IMAGE_ROOTFS_EXTRA_SPACE = "0"
BAD_RECOMMENDATIONS += "busybox-syslog busybox-udhcpc"

```

lyctw, 2024-08-29:
@kraj, thanks! 

I made an attempt and encountered this error.

```
[    4.622102] Kernel memory protection not selected by kernel config.
[    4.641179] Run /init as init process
[    4.654163] Failed to execute /init (error -13)
[    4.668554] Run /sbin/init as init process
[    4.683295] Run /etc/init as init process
[    4.697824] Run /bin/init as init process
[    4.712793] Run /bin/sh as init process
[    4.773827] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000000
[    4.796623] CPU: 1 PID: 1 Comm: sh Not tainted 6.1.47-yocto-standard-ae350 #1
[    4.817891] Hardware name: andestech,ax45 (DT)
[    4.831154] Call Trace:
[    4.838518] [<ffffffff80005b28>] dump_backtrace+0x1c/0x24
[    4.854749] [<ffffffff805fad9c>] show_stack+0x2c/0x38
[    4.869931] [<ffffffff806033c4>] dump_stack_lvl+0x3c/0x54
[    4.886124] [<ffffffff806033f0>] dump_stack+0x14/0x1c
[    4.901269] [<ffffffff805fb020>] panic+0xf2/0x276
[    4.915405] [<ffffffff80013de6>] do_exit+0x188/0x72e
[    4.930349] [<ffffffff8001454e>] sys_exit_group+0x0/0x18
[    4.946287] [<ffffffff80014566>] __wake_up_parent+0x0/0x20
[    4.962747] [<ffffffff800034f0>] ret_from_syscall+0x0/0x2
[    4.978922] SMP: stopping secondary CPUs
[    4.990872] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000000 ]---
```

kraj, 2024-08-29:
Perhaps some more options are needed for busybox - see https://github.com/YoeDistro/yoe-distro/blob/master/sources/meta-yoe/recipes-core/busybox/busybox/yoe_fragment.cfg

You can define busybox_%.bbappend in meta-riscv like this

https://github.com/YoeDistro/yoe-distro/blob/master/sources/meta-yoe/recipes-core/busybox/busybox_%25.bbappend

kraj, 2026-10-12:
closing for lack of response.

closed 2026-10-12 by kraj