riscv/meta-riscv #494: u-boot-beaglev-ahead.bb:do_configure failed

kanakshilledar, 2024-10-17:
**Description**
I am trying to build for yocto for BeagleV Ahead board and I am getting stuck on the following error.
```commandline
ERROR: Task (/home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/u-boot-beaglev-ahead.bb:do_configure) failed with exit code '1'
```
I am currently on Ubuntu 22.04 and using Yocto Scarthgap and master branch of **meta-riscv**.
I have set `MACHINE=beaglev-ahead`.

**Steps to reproduce the issue:**
1. Clone scarthgap of yocto.
2. Clone meta-riscv master branch
3. Create a build environment
4. Set `MACHINE=beaglev-ahead" in **conf/local.conf**
5. Run `bitbake core-image-minimal`

**Describe the results you received:**
The build will fail on `u-boot-beaglev-ahead.bb` recipe.
```console
 bitbake core-image-minimal
Loading cache: 100% |#################################################################| Time: 0:00:00
Loaded 1911 entries from dependency cache.
Parsing recipes: 100% |###############################################################| Time: 0:00:02
Parsing of 953 .bb files complete (950 cached, 3 parsed). 1914 targets, 100 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.8.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv64-poky-linux"
MACHINE              = "beaglev-ahead"
DISTRO               = "poky"
DISTRO_VERSION       = "5.0.4"
TUNE_FEATURES        = "riscv64"
meta                 
meta-poky            
meta-yocto-bsp       = "scarthgap:cd44e6bd40b0c1f498b3feaeb5e9b72f8bf32d41"
meta-riscv           = "master:e875ec652fecdfd3ae485d1226d29fbbcac33362"

Sstate summary: Wanted 689 Local 0 Mirrors 0 Missed 689 Current 1016 (0% match, 59% complete) 0:00:00
Initialising tasks: 100% |############################################################| Time: 0:00:03
NOTE: Executing Tasks
ERROR: u-boot-beaglev-ahead-1_1.0-r0 do_configure: ExecutionError('/home/kanak/yocto/build-beaglevahead/tmp/work/beaglev_ahead-poky-linux/u-boot-beaglev-ahead/1.0/temp/run.do_configure.2760807', 2, None, None)
ERROR: Logfile of failure stored in: /home/kanak/yocto/build-beaglevahead/tmp/work/beaglev_ahead-poky-linux/u-boot-beaglev-ahead/1.0/temp/log.do_configure.2760807
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/home/kanak/yocto/sources/poky/meta/recipes-core/glibc/glibc_2.39.bb:do_populate_sysroot', '/home/kanak/yocto/sources/poky/meta/recipes-devtools/gcc/gcc-cross_13.3.bb:do_populate_sysroot', '/home/kanak/yocto/sources/poky/meta/recipes-devtools/gcc/gcc-runtime_13.3.bb:do_populate_sysroot', '/home/kanak/yocto/sources/poky/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', '/home/kanak/yocto/sources/poky/meta/recipes-kernel/kern-tools/kern-tools-native_git.bb:do_populate_sysroot', 'virtual:native:/home/kanak/yocto/sources/poky/meta/recipes-bsp/u-boot/u-boot-tools_2024.01.bb:do_populate_sysroot', 'virtual:native:/home/kanak/yocto/sources/poky/meta/recipes-connectivity/openssl/openssl_3.2.3.bb:do_populate_sysroot', 'virtual:native:/home/kanak/yocto/sources/poky/meta/recipes-devtools/bison/bison_3.8.2.bb:do_populate_sysroot', 'virtual:native:/home/kanak/yocto/sources/poky/meta/recipes-devtools/flex/flex_2.6.4.bb:do_populate_sysroot', 'virtual:native:/home/kanak/yocto/sources/poky/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/home/kanak/yocto/sources/poky/meta/recipes-devtools/pkgconfig/pkgconfig_git.bb:do_populate_sysroot', 'virtual:native:/home/kanak/yocto/sources/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/home/kanak/yocto/sources/poky/meta/recipes-devtools/python/python3-setuptools_69.1.1.bb:do_populate_sysroot', 'virtual:native:/home/kanak/yocto/sources/poky/meta/recipes-devtools/python/python3_3.12.6.bb:do_populate_sysroot', 'virtual:native:/home/kanak/yocto/sources/poky/meta/recipes-devtools/swig/swig_4.2.1.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['gettext-minimal-native', 'glibc', 'binutils-cross-riscv64', 'cmake-native', 'gcc-cross-riscv64', 'gcc-runtime', 'libgcc', 'libtool-native', 'm4-native', 'quilt-native', 'texinfo-dummy-native', 'kern-tools-native', 'linux-libc-headers', 'u-boot-tools-native', 'openssl-native', 'expat-native', 'ncurses-native', 'util-linux-libuuid-native', 'util-linux-native', 'zlib-native', 'bison-native', 'flex-native', 'git-native', 'gnu-config-native', 'libedit-native', 'make-native', 'patch-native', 'perl-native', 'pkgconfig-native', 'pseudo-native', 'python3-build-native', 'python3-flit-core-native', 'python3-installer-native', 'python3-packaging-native', 'python3-pyproject-hooks-native', 'python3-setuptools-native', 'python3-wheel-native', 'python3-native', 'swig-native', 'bzip2-native', 'libidn2-native', 'libnsl2-native', 'libtirpc-native', 'unzip-native', 'xz-native', 'zstd-native', 'attr-native', 'ca-certificates-native', 'curl-native', 'gdbm-native', 'gmp-native', 'gnutls-native', 'libtasn1-native', 'libcap-ng-native', 'libffi-native', 'libmpc-native', 'libpcre2-native', 'libunistring-native', 'mpfr-native', 'nettle-native', 'sqlite3-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Searching for 0001-Set-sec_library-path-to-source-location.patch in paths:
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/u-boot-beaglev-ahead/poky
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/files/poky
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/u-boot-beaglev-ahead-1.0/poky
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/u-boot-beaglev-ahead/poky
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/files/poky
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/u-boot-beaglev-ahead/beaglev-ahead
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/files/beaglev-ahead
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/u-boot-beaglev-ahead-1.0/beaglev-ahead
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/u-boot-beaglev-ahead/beaglev-ahead
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/files/beaglev-ahead
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/u-boot-beaglev-ahead/c910
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/files/c910
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/u-boot-beaglev-ahead-1.0/c910
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/u-boot-beaglev-ahead/c910
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/files/c910
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/u-boot-beaglev-ahead/riscv64
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/files/riscv64
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/u-boot-beaglev-ahead-1.0/riscv64
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/u-boot-beaglev-ahead/riscv64
[... 81 lines cut ...]
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/files/riscv64
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/u-boot-beaglev-ahead/
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/files/
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/u-boot-beaglev-ahead-1.0/
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/u-boot-beaglev-ahead/
|     /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/files/
| DEBUG: Using /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/u-boot-beaglev-ahead/uEnv-beaglev-ahead.txt for uEnv-beaglev-ahead.txt
| DEBUG: Executing shell function do_configure
| sed: can't read /tftp-mmc-boot.txt: No such file or directory
| WARNING: exit code 2 from a shell command.
ERROR: Task (/home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/u-boot-beaglev-ahead.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2448 tasks of which 2322 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/u-boot-beaglev-ahead.bb:do_configure
Summary: There was 1 ERROR message, returning a non-zero exit code.
```
The full log of u-boot-beaglev-ahead is here: https://pastebin.com/dvVLadun

**Describe the results you expected:**
The build should pass

**Additional information you deem important (e.g. issue happens only occasionally):**
I tried doing `-c clean` and `-c cleansstate` and still this issue is there.
Thanks!

kraj, 2024-10-17:
> ```
> meta-yocto-bsp       = "scarthgap:cd44e6bd40b0c1f498b3feaeb5e9b72f8bf32d41"
> meta-riscv           = "master:e875ec652fecdfd3ae485d1226d29fbbcac33362"
> ```

@kanakshilledar how did you setup your project ? it seems base project is using `scarthgap` release but `meta-riscv` is at tip of `master` branch. I would suggest to use same branch for all the repos. If you want to stay with scarthgap then use scarthgap for meta-riscv as well

kanakshilledar, 2024-10-22:
Hi
@kraj sorry for the delay in response.
I tried to move both of them to master and now I am getting stuck at QA
```console
 bitbake core-image-minimal
Loading cache: 100% |#################################################################| Time: 0:00:00
Loaded 1912 entries from dependency cache.
Parsing recipes: 100% |###############################################################| Time: 0:00:02
Parsing of 952 .bb files complete (951 cached, 1 parsed). 1913 targets, 106 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.9.1"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv64-poky-linux"
MACHINE              = "beaglev-ahead"
DISTRO               = "poky"
DISTRO_VERSION       = "5.1"
TUNE_FEATURES        = "riscv64"
meta                 
meta-poky            
meta-yocto-bsp       = "master:c68ef7b4b7c732feb2d5f25aefd2c40ecf56e1a7"
meta-riscv           = "master:e875ec652fecdfd3ae485d1226d29fbbcac33362"

Sstate summary: Wanted 821 Local 0 Mirrors 0 Missed 821 Current 931 (0% match, 53% complete)  0:00:00
Initialising tasks: 100% |############################################################| Time: 0:00:01
NOTE: Executing Tasks
ERROR: linux-beaglev-dev-6.9.0+git-r0 do_patch: QA Issue: Missing Upstream-Status in patch
/home/kanak/yocto/sources/meta-riscv/recipes-kernel/linux/linux-beaglev-dev/0001-dt-bindings-reset-Document-th1520-reset-control.patch
Please add according to https://docs.yoctoproject.org/contributor-guide/recipe-style-guide.html#patch-upstream-status . [patch-status]
ERROR: linux-beaglev-dev-6.9.0+git-r0 do_patch: QA Issue: Missing Upstream-Status in patch
/home/kanak/yocto/sources/meta-riscv/recipes-kernel/linux/linux-beaglev-dev/0002-reset-Add-th1520-reset-driver-support.patch
Please add according to https://docs.yoctoproject.org/contributor-guide/recipe-style-guide.html#patch-upstream-status . [patch-status]
ERROR: linux-beaglev-dev-6.9.0+git-r0 do_patch: QA Issue: Missing Upstream-Status in patch
/home/kanak/yocto/sources/meta-riscv/recipes-kernel/linux/linux-beaglev-dev/0003-WIP-on-th_net-d3e1832c7d2d-riscv-mm-update-T-Head-me.patch
Please add according to https://docs.yoctoproject.org/contributor-guide/recipe-style-guide.html#patch-upstream-status . [patch-status]
ERROR: linux-beaglev-dev-6.9.0+git-r0 do_patch: QA Issue: Missing Upstream-Status in patch
/home/kanak/yocto/sources/meta-riscv/recipes-kernel/linux/linux-beaglev-dev/0004-dt-binding-riscv-add-T-HEAD-CPU-reset.patch
Please add according to https://docs.yoctoproject.org/contributor-guide/recipe-style-guide.html#patch-upstream-status . [patch-status]
ERROR: linux-beaglev-dev-6.9.0+git-r0 do_patch: QA Issue: Missing Upstream-Status in patch
/home/kanak/yocto/sources/meta-riscv/recipes-kernel/linux/linux-beaglev-dev/0005-th1520-add-cpu-reset-node.patch
Please add according to https://docs.yoctoproject.org/contributor-guide/recipe-style-guide.html#patch-upstream-status . [patch-status]
ERROR: linux-beaglev-dev-6.9.0+git-r0 do_patch: QA Issue: Missing Upstream-Status in patch
/home/kanak/yocto/sources/meta-riscv/recipes-kernel/linux/linux-beaglev-dev/0006-th1520-beaglev-ahead.dts-enable-ethernet0.patch
Please add according to https://docs.yoctoproject.org/contributor-guide/recipe-style-guide.html#patch-upstream-status . [patch-status]
ERROR: linux-beaglev-dev-6.9.0+git-r0 do_patch: QA Issue: Missing Upstream-Status in patch
/home/kanak/yocto/sources/meta-riscv/recipes-kernel/linux/linux-beaglev-dev/0007-v6.8-rc3-fix-reset-th1520.c-build.patch
Please add according to https://docs.yoctoproject.org/contributor-guide/recipe-style-guide.html#patch-upstream-status . [patch-status]
ERROR: linux-beaglev-dev-6.9.0+git-r0 do_patch: Fatal QA errors were found, failing task.
ERROR: Logfile of failure stored in: /home/kanak/yocto/buildtead/tmp/work/beaglev_ahead-poky-linux/linux-beaglev-dev/6.9.0+git/temp/log.do_patch.912973
ERROR: Task (/home/kanak/yocto/sources/meta-riscv/recipes-kernel/linux/linux-beaglev-dev.bb:do_patch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2093 tasks of which 2087 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/kanak/yocto/sources/meta-riscv/recipes-kernel/linux/linux-beaglev-dev.bb:do_patch
    log: /home/kanak/yocto/buildtead/tmp/work/beaglev_ahead-poky-linux/linux-beaglev-dev/6.9.0+git/temp/log.do_patch.912973
Summary: There were 8 ERROR messages, returning a non-zero exit code.
```
The full log can be found here: https://pastebin.com/Y4esM14d

Shall I add the `Upstream-Status: Pending` in these patches?
As per this documentation: https://docs.yoctoproject.org/contributor-guide/recipe-style-guide.html#patch-upstream-status

Thanks!

kraj, 2024-10-22:
> Hi @kraj sorry for the delay in response. I tried to move both of them to master and now I am getting stuck at QA
> 
> ```
>  bitbake core-image-minimal
> Loading cache: 100% |#################################################################| Time: 0:00:00
> Loaded 1912 entries from dependency cache.
> Parsing recipes: 100% |###############################################################| Time: 0:00:02
> Parsing of 952 .bb files complete (951 cached, 1 parsed). 1913 targets, 106 skipped, 0 masked, 0 errors.
> NOTE: Resolving any missing task queue dependencies
> 
> Build Configuration:
> BB_VERSION           = "2.9.1"
> BUILD_SYS            = "x86_64-linux"
> NATIVELSBSTRING      = "universal"
> TARGET_SYS           = "riscv64-poky-linux"
> MACHINE              = "beaglev-ahead"
> DISTRO               = "poky"
> DISTRO_VERSION       = "5.1"
> TUNE_FEATURES        = "riscv64"
> meta                 
> meta-poky            
> meta-yocto-bsp       = "master:c68ef7b4b7c732feb2d5f25aefd2c40ecf56e1a7"
> meta-riscv           = "master:e875ec652fecdfd3ae485d1226d29fbbcac33362"
> 
> Sstate summary: Wanted 821 Local 0 Mirrors 0 Missed 821 Current 931 (0% match, 53% complete)  0:00:00
> Initialising tasks: 100% |############################################################| Time: 0:00:01
> NOTE: Executing Tasks
> ERROR: linux-beaglev-dev-6.9.0+git-r0 do_patch: QA Issue: Missing Upstream-Status in patch
> /home/kanak/yocto/sources/meta-riscv/recipes-kernel/linux/linux-beaglev-dev/0001-dt-bindings-reset-Document-th1520-reset-control.patch
> Please add according to https://docs.yoctoproject.org/contributor-guide/recipe-style-guide.html#patch-upstream-status . [patch-status]
> ERROR: linux-beaglev-dev-6.9.0+git-r0 do_patch: QA Issue: Missing Upstream-Status in patch
> /home/kanak/yocto/sources/meta-riscv/recipes-kernel/linux/linux-beaglev-dev/0002-reset-Add-th1520-reset-driver-support.patch
> Please add according to https://docs.yoctoproject.org/contributor-guide/recipe-style-guide.html#patch-upstream-status . [patch-status]
> ERROR: linux-beaglev-dev-6.9.0+git-r0 do_patch: QA Issue: Missing Upstream-Status in patch
> /home/kanak/yocto/sources/meta-riscv/recipes-kernel/linux/linux-beaglev-dev/0003-WIP-on-th_net-d3e1832c7d2d-riscv-mm-update-T-Head-me.patch
> Please add according to https://docs.yoctoproject.org/contributor-guide/recipe-style-guide.html#patch-upstream-status . [patch-status]
> ERROR: linux-beaglev-dev-6.9.0+git-r0 do_patch: QA Issue: Missing Upstream-Status in patch
> /home/kanak/yocto/sources/meta-riscv/recipes-kernel/linux/linux-beaglev-dev/0004-dt-binding-riscv-add-T-HEAD-CPU-reset.patch
> Please add according to https://docs.yoctoproject.org/contributor-guide/recipe-style-guide.html#patch-upstream-status . [patch-status]
> ERROR: linux-beaglev-dev-6.9.0+git-r0 do_patch: QA Issue: Missing Upstream-Status in patch
> /home/kanak/yocto/sources/meta-riscv/recipes-kernel/linux/linux-beaglev-dev/0005-th1520-add-cpu-reset-node.patch
> Please add according to https://docs.yoctoproject.org/contributor-guide/recipe-style-guide.html#patch-upstream-status . [patch-status]
> ERROR: linux-beaglev-dev-6.9.0+git-r0 do_patch: QA Issue: Missing Upstream-Status in patch
> /home/kanak/yocto/sources/meta-riscv/recipes-kernel/linux/linux-beaglev-dev/0006-th1520-beaglev-ahead.dts-enable-ethernet0.patch
> Please add according to https://docs.yoctoproject.org/contributor-guide/recipe-style-guide.html#patch-upstream-status . [patch-status]
> ERROR: linux-beaglev-dev-6.9.0+git-r0 do_patch: QA Issue: Missing Upstream-Status in patch
> /home/kanak/yocto/sources/meta-riscv/recipes-kernel/linux/linux-beaglev-dev/0007-v6.8-rc3-fix-reset-th1520.c-build.patch
> Please add according to https://docs.yoctoproject.org/contributor-guide/recipe-style-guide.html#patch-upstream-status . [patch-status]
> ERROR: linux-beaglev-dev-6.9.0+git-r0 do_patch: Fatal QA errors were found, failing task.
> ERROR: Logfile of failure stored in: /home/kanak/yocto/buildtead/tmp/work/beaglev_ahead-poky-linux/linux-beaglev-dev/6.9.0+git/temp/log.do_patch.912973
> ERROR: Task (/home/kanak/yocto/sources/meta-riscv/recipes-kernel/linux/linux-beaglev-dev.bb:do_patch) failed with exit code '1'
> NOTE: Tasks Summary: Attempted 2093 tasks of which 2087 didn't need to be rerun and 1 failed.
> 
> Summary: 1 task failed:
>   /home/kanak/yocto/sources/meta-riscv/recipes-kernel/linux/linux-beaglev-dev.bb:do_patch
>     log: /home/kanak/yocto/buildtead/tmp/work/beaglev_ahead-poky-linux/linux-beaglev-dev/6.9.0+git/temp/log.do_patch.912973
> Summary: There were 8 ERROR messages, returning a non-zero exit code.
> ```
> 
> The full log can be found here: https://pastebin.com/Y4esM14d
> 
> Shall I add the `Upstream-Status: Pending` in these patches? As per this documentation: https://docs.yoctoproject.org/contributor-guide/recipe-style-guide.html#patch-upstream-status
> 
> Thanks!

that a problem in meta-riscv, and yes adding Upstream-Status is needed. Something like
https://github.com/riscv/meta-riscv/pull/496

kanakshilledar, 2024-10-24:
Thanks I will try to add these and try to create a PR for unfixed parts. Can I get this issue assigned to me?
I am getting similar error in `u-boot-beaglev-ahead`
```console
ERROR: u-boot-beaglev-ahead-1_1.0-r0 do_patch: QA Issue: Missing Upstream-Status in patch
/home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/u-boot-beaglev-ahead/0001-Set-sec_library-path-to-source-location.patch
Please add according to https://docs.yoctoproject.org/contributor-guide/recipe-style-guide.html#patch-upstream-status . [patch-status]
ERROR: u-boot-beaglev-ahead-1_1.0-r0 do_patch: QA Issue: Missing Upstream-Status in patch
/home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/u-boot-beaglev-ahead/0002-Fix-errors-for-GCC-14-compatibility.patch
Please add according to https://docs.yoctoproject.org/contributor-guide/recipe-style-guide.html#patch-upstream-status . [patch-status]
ERROR: u-boot-beaglev-ahead-1_1.0-r0 do_patch: Fatal QA errors were found, failing task.
ERROR: Logfile of failure stored in: /home/kanak/yocto/buildtead/tmp/work/beaglev_ahead-poky-linux/u-boot-beaglev-ahead/1.0/temp/log.do_patch.2994457
ERROR: Task (/home/kanak/yocto/sources/meta-riscv/recipes-bsp/u-boot/u-boot-beaglev-ahead.bb:do_patch) failed with exit code '1'
````

kraj, 2024-10-24:
@kanakshilledar please go ahead and send a PR

PR #497 (2024-10-25): fix u-boot-beaglev-ahead missing upstream status

`MACHINE = "beaglev-ahead"`: Added upstream status to u-boot for beaglev ahead.
`busybox`: Bumped up the bbappend version of busybox.

This PR closes #494
closed 2024-10-25 by kraj