riscv/meta-riscv #502: gcc-cross-riscv64-14.2.0-r0 hangs at do_compile()

maroueneboubakri, 2024-11-03:
Hello,

Trying instruction on README file, with 2 machines with 2 different configs (Ubuntu 22 and Ubuntu 20), gcc-cross-riscv64 always hangs at do_compile().

Best
Maro

maroueneboubakri, 2024-11-03:
FI
```
| g++   -fno-PIE -c   -isystem/home/maro/riscv/riscv-yocto/build/tmp/work/x86_64-linux/gcc-cross-riscv64/14.2.0/recipe-sysroot-native/usr/include -O2 -pipe   -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -fno-PIE -I. -I. -I../../../../../../../work-shared/gcc-14.2.0-r0/gcc-14.2.0/gcc -I../../../../../../../work-shared/gcc-14.2.0-r0/gcc-14.2.0/gcc/. -I../../../../../../../work-shared/gcc-14.2.0-r0/gcc-14.2.0/gcc/../include  -I../../../../../../../work-shared/gcc-14.2.0-r0/gcc-14.2.0/gcc/../libcpp/include -I../../../../../../../work-shared/gcc-14.2.0-r0/gcc-14.2.0/gcc/../libcody  -I../../../../../../../work-shared/gcc-14.2.0-r0/gcc-14.2.0/gcc/../libdecnumber -I../../../../../../../work-shared/gcc-14.2.0-r0/gcc-14.2.0/gcc/../libdecnumber/dpd -I../libdecnumber -I../../../../../../../work-shared/gcc-14.2.0-r0/gcc-14.2.0/gcc/../libbacktrace   -o ggc-page.o -MT ggc-page.o -MMD -MP -MF ./.deps/ggc-page.TPo ../../../../../../../work-shared/gcc-14.2.0-r0/gcc-14.2.0/gcc/ggc-page.cc
| {standard input}: Assembler messages:
| {standard input}:74061: Warning: end of file not at end of a line; newline inserted
| {standard input}:74105: Error: unrecognized symbol type ""
| g++: fatal error: Killed signal terminated program cc1plus
| compilation terminated.
| make[1]: *** [Makefile:1199: insn-emit-2.o] Error 1
| make[1]: *** Waiting for unfinished jobs....
| /bin/bash ../../../../../../../work-shared/gcc-14.2.0-r0/gcc-14.2.0/gcc/../move-if-change tmp-output.cc insn-output.cc
| echo timestamp > s-output
| rm gcc.pod
| make[1]: Leaving directory '/home/maro/riscv/riscv-yocto/build/tmp/work/x86_64-linux/gcc-cross-riscv64/14.2.0/gcc-14.2.0/build.x86_64-linux.riscv64-poky-linux/gcc'
| make: *** [Makefile:4699: all-gcc] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/maro/riscv/riscv-yocto/poky/meta/recipes-devtools/gcc/gcc-cross_14.2.bb:do_compile) failed with exit code '1'
```

kraj, 2024-11-03:
@maroueneboubakri it seems gcc is not having enough memory when its compiling gcc-cross, How much memory do you have on build system ? 

Maybe try to reduce parallism by setting 

```
PARALLEL_MAKE:append = " -l ${@int(os.sysconf(os.sysconf_names['SC_NPROCESSORS_ONLN']) * 100/100)}"
```

maroueneboubakri, 2024-11-03:
> @maroueneboubakri it seems gcc is not having enough memory when its compiling gcc-cross, How much memory do you have on build system ?
> 
> Maybe try to reduce parallism by setting it in `conf/local.conf`
> 
> ```
> PARALLEL_MAKE:append = " -l ${@int(os.sysconf(os.sysconf_names['SC_NPROCESSORS_ONLN']) * 100/100)}"
> ```

@kraj thank you for support, my machine has 16GB RAM. Reducing number of jobs solved issue with compiling gcc ! thank you, however llvm fails with the following error:

``` 
| [1079/2598] cd /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/build && /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/recipe-sysroot-native/usr/bin/llvm-tblgen19.1.2 -gen-disassembler -I /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/llvm-project-19.1.2.src/llvm/lib/Target/AMDGPU -I/home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/build/include -I/home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/llvm-project-19.1.2.src/llvm/include -I /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/llvm-project-19.1.2.src/llvm/lib/Target /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/llvm-project-19.1.2.src/llvm/lib/Target/AMDGPU/AMDGPU.td --write-if-changed -o lib/Target/AMDGPU/AMDGPUGenDisassemblerTables.inc -d lib/Target/AMDGPU/AMDGPUGenDisassemblerTables.inc.d
| [1080/2598] cd /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/build && /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/recipe-sysroot-native/usr/bin/llvm-tblgen19.1.2 -gen-global-isel -I /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/llvm-project-19.1.2.src/llvm/lib/Target/AMDGPU -I/home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/build/include -I/home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/llvm-project-19.1.2.src/llvm/include -I /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/llvm-project-19.1.2.src/llvm/lib/Target /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/llvm-project-19.1.2.src/llvm/lib/Target/AMDGPU/AMDGPUGISel.td --write-if-changed -o lib/Target/AMDGPU/AMDGPUGenGlobalISel.inc -d lib/Target/AMDGPU/AMDGPUGenGlobalISel.inc.d
| [1081/2598] cd /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/build && /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/recipe-sysroot-native/usr/bin/llvm-tblgen19.1.2 -gen-dag-isel -I /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/llvm-project-19.1.2.src/llvm/lib/Target/AMDGPU -I/home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/build/include -I/home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/llvm-project-19.1.2.src/llvm/include -I /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/llvm-project-19.1.2.src/llvm/lib/Target -omit-comments /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/llvm-project-19.1.2.src/llvm/lib/Target/AMDGPU/AMDGPU.td --write-if-changed -o lib/Target/AMDGPU/AMDGPUGenDAGISel.inc -d lib/Target/AMDGPU/AMDGPUGenDAGISel.inc.d
| [1082/2598] cd /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/build && /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/recipe-sysroot-native/usr/bin/llvm-tblgen19.1.2 -gen-asm-writer -I /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/llvm-project-19.1.2.src/llvm/lib/Target/AMDGPU -I/home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/build/include -I/home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/llvm-project-19.1.2.src/llvm/include -I /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/llvm-project-19.1.2.src/llvm/lib/Target /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/llvm-project-19.1.2.src/llvm/lib/Target/AMDGPU/AMDGPU.td --write-if-changed -o lib/Target/AMDGPU/AMDGPUGenAsmWriter.inc -d lib/Target/AMDGPU/AMDGPUGenAsmWriter.inc.d
| [1083/2598] cd /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/build && /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/recipe-sysroot-native/usr/bin/llvm-tblgen19.1.2 -gen-asm-matcher -I /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/llvm-project-19.1.2.src/llvm/lib/Target/AMDGPU -I/home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/build/include -I/home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/llvm-project-19.1.2.src/llvm/include -I /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/llvm-project-19.1.2.src/llvm/lib/Target /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/llvm-project-19.1.2.src/llvm/lib/Target/AMDGPU/AMDGPU.td --write-if-changed -o lib/Target/AMDGPU/AMDGPUGenAsmMatcher.inc -d lib/Target/AMDGPU/AMDGPUGenAsmMatcher.inc.d
| ninja: build stopped: subcommand failed.
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/maro/riscv/riscv-yocto/poky/meta/recipes-devtools/llvm/llvm_19.1.2.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 780 tasks of which 634 didn't need to be rerun and 1 failed.
[... cut ...]
NOTE:   do_populate_sysroot: 0.0% sstate reuse(0 setscene, 23 scratch)
NOTE:   do_deploy_source_date_epoch: 0.0% sstate reuse(0 setscene, 1 scratch)
NOTE:   do_recipe_qa: 0.0% sstate reuse(0 setscene, 1 scratch)

Summary: 1 task failed:
  /home/maro/riscv/riscv-yocto/poky/meta/recipes-devtools/llvm/llvm_19.1.2.bb:do_compile
    log: /home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/llvm/19.1.2/temp/log.do_compile.12215
Summary: There was 1 WARNING message.
Summary: There was 1 ERROR message, returning a non-zero exit code.
```

maroueneboubakri, 2024-11-03:
More, while compiling gcc, (not gcc-corss above)
```
| riscv64-poky-linux-g++     -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2  -O2 -g --sysroot=/home/nxf45729/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/gcc/14.2.0/recipe-sysroot  -fno-PIE -c   -O2 -g -fcanon-prefix-map  -fmacro-prefix-map=/home/nxf45729/riscv/riscv-yocto/build/tmp/work-shared/gcc-14.2.0-r0/gcc-14.2.0=/usr/src/debug/gcc/14.2.0  -fdebug-prefix-map=/home/nxf45729/riscv/riscv-yocto/build/tmp/work-shared/gcc-14.2.0-r0/gcc-14.2.0=/usr/src/debug/gcc/14.2.0  -fmacro-prefix-map=/home/nxf45729/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/gcc/14.2.0/gcc-14.2.0/build.riscv64-poky-linux.riscv64-poky-linux=/usr/src/debug/gcc/14.2.0  -fdebug-prefix-map=/home/nxf45729/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/gcc/14.2.0/gcc-14.2.0/build.riscv64-poky-linux.riscv64-poky-linux=/usr/src/debug/gcc/14.2.0  -fdebug-prefix-map=/home/nxf45729/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/gcc/14.2.0/recipe-sysroot=  -fmacro-prefix-map=/home/nxf45729/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/gcc/14.2.0/recipe-sysroot=  -fdebug-prefix-map=/home/nxf45729/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/gcc/14.2.0/recipe-sysroot-native=  -fmacro-prefix-map=/home/nxf45729/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/gcc/14.2.0/recipe-sysroot-native=  -pipe -fvisibility-inlines-hidden   -DIN_GCC    -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -fno-PIE -I. -I. -I../../../../../../../work-shared/gcc-14.2.0-r0/gcc-14.2.0/gcc -I../../../../../../../work-shared/gcc-14.2.0-r0/gcc-14.2.0/gcc/. -I../../../../../../../work-shared/gcc-14.2.0-r0/gcc-14.2.0/gcc/../include  -I../../../../../../../work-shared/gcc-14.2.0-r0/gcc-14.2.0/gcc/../libcpp/include -I../../../../../../../work-shared/gcc-14.2.0-r0/gcc-14.2.0/gcc/../libcody  -I../../../../../../../work-shared/gcc-14.2.0-r0/gcc-14.2.0/gcc/../libdecnumber -I../../../../../../../work-shared/gcc-14.2.0-r0/gcc-14.2.0/gcc/../libdecnumber/dpd -I../libdecnumber -I../../../../../../../work-shared/gcc-14.2.0-r0/gcc-14.2.0/gcc/../libbacktrace   -o insn-latencytab.o -MT insn-latencytab.o -MMD -MP -MF ./.deps/insn-latencytab.TPo insn-latencytab.cc
| {standard input}: Assembler messages:
| {standard input}: Error: open CFI at the end of file; missing .cfi_endproc directive
| riscv64-poky-linux-g++: fatal error: Killed signal terminated program cc1plus
| compilation terminated.
| make[1]: *** [Makefile:1199: insn-emit-7.o] Error 1
| make[1]: *** Waiting for unfinished jobs....
| rm gcc.pod
| make[1]: Leaving directory '/home/nxf45729/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/gcc/14.2.0/gcc-14.2.0/build.riscv64-poky-linux.riscv64-poky-linux/gcc'
| make: *** [Makefile:4699: all-gcc] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/nxf45729/riscv/riscv-yocto/poky/meta/recipes-devtools/gcc/gcc_14.2.bb:do_compile) failed with exit code '1'
Waiting for 1 running task to finish:
```
monitoring the memory/cpu consumption seems fine, no memory overrun though.

maroueneboubakri, 2024-11-03:
Not able to link cc1plus:
I switched to another machine and same issue occurs.
 
```
| riscv64-poky-linux-g++     -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2  -O2 -g --sysroot=/home/maro/riscv/riscv-yo                  cto/build/tmp/work/riscv64-poky-linux/gcc/14.2.0/recipe-sysroot -no-pie   -O2 -g -fcanon-prefix-map  -fmacro-prefix-map=/home/                  maro/riscv/riscv-yocto/build/tmp/work-shared/gcc-14.2.0-r0/gcc-14.2.0=/usr/src/debug/gcc/14.2.0  -fdebug-prefix-map=/home/                  maro/riscv/riscv-yocto/build/tmp/work-shared/gcc-14.2.0-r0/gcc-14.2.0=/usr/src/debug/gcc/14.2.0  -fmacro-prefix-map=/home/                  maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/gcc/14.2.0/gcc-14.2.0/build.riscv64-poky-linux.riscv64-poky-linux                  =/usr/src/debug/gcc/14.2.0  -fdebug-prefix-map=/home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/gcc/14.2.0/g                  cc-14.2.0/build.riscv64-poky-linux.riscv64-poky-linux=/usr/src/debug/gcc/14.2.0  -fdebug-prefix-map=/home/maro/riscv/riscv                  -yocto/build/tmp/work/riscv64-poky-linux/gcc/14.2.0/recipe-sysroot=  -fmacro-prefix-map=/home/maro/riscv/riscv-yocto/build                  /tmp/work/riscv64-poky-linux/gcc/14.2.0/recipe-sysroot=  -fdebug-prefix-map=/home/maro/riscv/riscv-yocto/build/tmp/work/ri                  scv64-poky-linux/gcc/14.2.0/recipe-sysroot-native=  -fmacro-prefix-map=/home/maro/riscv/riscv-yocto/build/tmp/work/riscv64                  -poky-linux/gcc/14.2.0/recipe-sysroot-native=  -pipe -fvisibility-inlines-hidden   -DIN_GCC    -fno-exceptions -fno-rtti -fasy                  nchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-support                  ed -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -no-pie -stati                  c-libstdc++ -static-libgcc -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fcanon-prefix-map  -fmacro-prefix-map=/home/maro/                  riscv/riscv-yocto/build/tmp/work-shared/gcc-14.2.0-r0/gcc-14.2.0=/usr/src/debug/gcc/14.2.0  -fdebug-prefix-map=/home/maro/                  riscv/riscv-yocto/build/tmp/work-shared/gcc-14.2.0-r0/gcc-14.2.0=/usr/src/debug/gcc/14.2.0  -fmacro-prefix-map=/home/maro/                  riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/gcc/14.2.0/gcc-14.2.0/build.riscv64-poky-linux.riscv64-poky-linux=/usr/src                  /debug/gcc/14.2.0  -fdebug-prefix-map=/home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/gcc/14.2.0/gcc-14.2.0                  /build.riscv64-poky-linux.riscv64-poky-linux=/usr/src/debug/gcc/14.2.0  -fdebug-prefix-map=/home/maro/riscv/riscv-yocto/bu                  ild/tmp/work/riscv64-poky-linux/gcc/14.2.0/recipe-sysroot=  -fmacro-prefix-map=/home/maro/riscv/riscv-yocto/build/tmp/work                  /riscv64-poky-linux/gcc/14.2.0/recipe-sysroot=  -fdebug-prefix-map=/home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-pok                  y-linux/gcc/14.2.0/recipe-sysroot-native=  -fmacro-prefix-map=/home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-lin                  ux/gcc/14.2.0/recipe-sysroot-native=  -Wl,-z,relro,-z,now -o cc1plus \
|       cp/cp-lang.o c-family/stub-objc.o cp/call.o cp/class.o cp/constexpr.o cp/constraint.o cp/coroutines.o cp/cp-gimplify.o                   cp/cp-objcp-common.o cp/cp-ubsan.o cp/cvt.o cp/contracts.o cp/cxx-pretty-print.o cp/decl.o cp/decl2.o cp/dump.o cp/error.o cp                  /except.o cp/expr.o cp/friend.o cp/init.o cp/lambda.o cp/lex.o cp/logic.o cp/mangle.o cp/mapper-client.o cp/mapper-resolver.o                   cp/method.o cp/module.o cp/name-lookup.o cp/optimize.o cp/parser.o cp/pt.o cp/ptree.o cp/rtti.o cp/search.o cp/semantics.o cp/                  tree.o cp/typeck.o cp/typeck2.o cp/vtable-class-hierarchy.o attribs.o c-family/c-common.o c-family/c-cppbuiltin.o c-family/c-d                  ump.o c-family/c-format.o c-family/c-gimplify.o c-family/c-indentation.o c-family/c-lex.o c-family/c-omp.o c-family/c-opts.o c                  -family/c-pch.o c-family/c-ppoutput.o c-family/c-pragma.o c-family/c-pretty-print.o c-family/c-semantics.o c-family/c-ada-spec                  .o c-family/c-ubsan.o c-family/known-headers.o c-family/c-attribs.o c-family/c-warn.o c-family/c-spellcheck.o glibc-c.o cc1plu                  s-checksum.o libbackend.a main.o libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a ../libcody/l                  ibcody.a  \
|       libcommon.a ../libcpp/libcpp.a   ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecn                  umber.a   -lmpc -lmpfr -lgmp -rdynamic  -lz -lzstd
| collect2: fatal error: ld terminated with signal 9 [Killed]
| compilation terminated.
| make[1]: *** [../../../../../../../work-shared/gcc-14.2.0-r0/gcc-14.2.0/gcc/cp/Make-lang.in:145: cc1plus] Error 1
| make[1]: *** Waiting for unfinished jobs....
| collect2: fatal error: ld terminated with signal 9 [Killed]
| compilation terminated.
| make[1]: *** [../../../../../../../work-shared/gcc-14.2.0-r0/gcc-14.2.0/gcc/c/Make-lang.in:87: cc1] Error 1
| make[1]: Leaving directory '/home/maro/riscv/riscv-yocto/build/tmp/work/riscv64-poky-linux/gcc/14.2.0/gcc-14.2.0/build.r                  iscv64-poky-linux.riscv64-poky-linux/gcc'
| make: *** [Makefile:4699: all-gcc] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
```

maroueneboubakri, 2024-11-03:
@kraj Unfortunately, it does not work, even reducing to single thread, Is there any CI in place for the project ?

Here is a full log:

```
maro@SMW002635:~/riscv/riscv-yocto/build$
maro@SMW002635:~/riscv/riscv-yocto/build$ nano conf/local.conf
maro@SMW002635:~/riscv/riscv-yocto/build$ MACHINE=qemuriscv64 bitbake -v core-image-full-cmdline
Loading cache: 100% |############################################################################################################| Time: 0:00:01
Loaded 4856 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies
NOTE: selecting pseudo-native to satisfy virtual/fakeroot-native due to PREFERRED_PROVIDERS
NOTE: selecting linux-yocto to satisfy virtual/kernel due to PREFERRED_PROVIDERS
NOTE: selecting opkg-utils-native to satisfy virtual/update-alternatives-native due to PREFERRED_PROVIDERS
NOTE: selecting opkg-native to satisfy opkg-native due to PREFERRED_PROVIDERS
NOTE: selecting binutils-cross-riscv64 to satisfy virtual/riscv64-poky-linux-binutils due to PREFERRED_PROVIDERS
NOTE: selecting opkg to satisfy runtime opkg due to PREFERRED_PROVIDER_opkg = opkg
NOTE: selecting glibc-locale to satisfy runtime locale-base-c due to PREFERRED_PROVIDER_virtual/libc-locale = glibc-locale
NOTE: selecting glibc-locale to satisfy runtime locale-base-en-gb due to PREFERRED_PROVIDER_virtual/libc-locale = glibc-locale
NOTE: selecting glibc-locale to satisfy runtime locale-base-en-us due to PREFERRED_PROVIDER_virtual/libc-locale = glibc-locale
NOTE: selecting linux-yocto to satisfy runtime kernel-base due to PREFERRED_PROVIDER_virtual/kernel = linux-yocto
NOTE: selecting linux-yocto to satisfy runtime kernel-image-image due to PREFERRED_PROVIDER_virtual/kernel = linux-yocto
NOTE: selecting linux-yocto to satisfy runtime kernel-image-uimage due to PREFERRED_PROVIDER_virtual/kernel = linux-yocto
NOTE: selecting linux-yocto to satisfy runtime kernel-image due to PREFERRED_PROVIDER_virtual/kernel = linux-yocto
NOTE: selecting u-boot to satisfy runtime u-boot-dev due to PREFERRED_PROVIDER_virtual/bootloader = u-boot
NOTE: selecting u-boot to satisfy runtime u-boot-env due to PREFERRED_PROVIDER_virtual/bootloader = u-boot
NOTE: selecting u-boot to satisfy runtime u-boot due to PREFERRED_PROVIDER_virtual/bootloader = u-boot
NOTE: selecting pkgconfig-native to satisfy pkgconfig-native due to PREFERRED_PROVIDERS
NOTE: selecting pigz-native to satisfy gzip-native due to PREFERRED_PROVIDERS
NOTE: selecting gcc-cross-riscv64 to satisfy virtual/riscv64-poky-linux-gcc due to PREFERRED_PROVIDERS
NOTE: selecting openssl-native to satisfy openssl-native due to PREFERRED_PROVIDERS
NOTE: selecting binutils-crosssdk-x86_64-pokysdk-linux to satisfy virtual/x86_64-pokysdk-linux-binutils due to PREFERRED_PROVIDERS
NOTE: selecting nativesdk-glibc to satisfy virtual/nativesdk-libc due to PREFERRED_PROVIDERS
NOTE: selecting gcc-crosssdk-x86_64-pokysdk-linux to satisfy virtual/x86_64-pokysdk-linux-gcc due to PREFERRED_PROVIDERS
NOTE: selecting glibc to satisfy virtual/libc due to PREFERRED_PROVIDERS
NOTE: selecting gcc-runtime to satisfy virtual/riscv64-poky-linux-compilerlibs due to PREFERRED_PROVIDERS
NOTE: selecting opkg-utils to satisfy virtual/update-alternatives due to PREFERRED_PROVIDERS
NOTE: selecting nativesdk-openssl to satisfy nativesdk-openssl due to PREFERRED_PROVIDERS
NOTE: selecting nativesdk-glibc to satisfy virtual/nativesdk-libintl due to PREFERRED_PROVIDERS
NOTE: selecting nativesdk-glibc to satisfy virtual/nativesdk-libiconv due to PREFERRED_PROVIDERS
NOTE: selecting glibc to satisfy runtime glibc-thread-db due to PREFERRED_PROVIDER_virtual/libc = glibc
NOTE: selecting glibc-locale to satisfy runtime glibc-gconv-iso8859-1 due to PREFERRED_PROVIDER_virtual/libc-locale = glibc-locale
NOTE: selecting gcc-runtime to satisfy runtime libstdc++-dev due to PREFERRED_PROVIDER_virtual/riscv64-poky-linux-compilerlibs = gcc-runtime
NOTE: selecting glibc to satisfy runtime glibc due to PREFERRED_PROVIDER_virtual/libc = glibc
NOTE: selecting gcc-runtime to satisfy runtime libatomic due to PREFERRED_PROVIDER_virtual/riscv64-poky-linux-compilerlibs = gcc-runtime
NOTE: selecting glibc-locale to satisfy runtime glibc-gconv-ibm850 due to PREFERRED_PROVIDER_virtual/libc-locale = glibc-locale
NOTE: selecting libgcc to satisfy runtime libgcc due to PREFERRED_PROVIDER_libgcc = libgcc
NOTE: selecting glibc to satisfy runtime glibc-dev due to PREFERRED_PROVIDER_virtual/libc = glibc
NOTE: selecting glibc-locale to satisfy runtime glibc-gconv-cp1252 due to PREFERRED_PROVIDER_virtual/libc-locale = glibc-locale
NOTE: selecting glibc to satisfy runtime glibc-utils due to PREFERRED_PROVIDER_virtual/libc = glibc
NOTE: selecting gcc-runtime to satisfy runtime libatomic-dev due to PREFERRED_PROVIDER_virtual/riscv64-poky-linux-compilerlibs = gcc-runtime
NOTE: selecting glibc-locale to satisfy runtime glibc-localedata-i18n due to PREFERRED_PROVIDER_virtual/libc-locale = glibc-locale
NOTE: selecting glibc to satisfy runtime glibc-dbg due to PREFERRED_PROVIDER_virtual/libc = glibc
NOTE: selecting glibc-locale to satisfy runtime glibc-gconv-iso8859-15 due to PREFERRED_PROVIDER_virtual/libc-locale = glibc-locale
NOTE: selecting gcc-runtime to satisfy runtime libstdc++ due to PREFERRED_PROVIDER_virtual/riscv64-poky-linux-compilerlibs = gcc-runtime
NOTE: selecting libgcc to satisfy runtime libgcc-dev due to PREFERRED_PROVIDER_libgcc = libgcc
[... cut ...]
echo timestamp > s-fixinc_list

collect2: fatal error: ld terminated with signal 9 [Killed]
compilation terminated.

make[1]: *** [../../../../../../../work-shared/gcc-14.2.0-r0/gcc-14.2.0/gcc/cp/Make-lang.in:145: cc1plus] Error 1
make[1]: *** Waiting for unfinished jobs....

Setscene tasks: 2862 of 2862
Currently  1 running tasks (3666 of 5954)  61% |##########################################################                                     |
0: gcc-14.2.0-r0 do_compile - 38s (pid 16097)
```

maroueneboubakri, 2024-11-03:
@kraj was able to build it 
```
BB_NUMBER_THREADS = "1"
PARALLEL_MAKE = "-j 1"
```
Thank you for your help, that would be good to have such info in the README file despite that those are obvious Yocto stuff. But I have the intention that this taking much memory compared to a Yocto build for ARM64.

Closing the thread.

closed 2024-11-03 by maroueneboubakri

kraj, 2024-11-03:
> @kraj was able to build it
> 
> ```
> BB_NUMBER_THREADS = "1"
> PARALLEL_MAKE = "-j 1"
> ```
> 
> Thank you for your help, that would be good to have such info in the README file despite that those are obvious Yocto stuff. But I have the intention that this taking much memory compared to a Yocto build for ARM64.

that is interesting observation, and its possible that RISCV64 cross target in gcc does have memory issues.
Its not documented because usually defaults work out ok. 

> 
> Closing the thread.

Glad, its working now, although -j1 and single thread is not ideal. Usually in latest yocto bitbake there is a pressure monitor which takes care of not overloading the system and adjusts issues new tasks based upon load ratio.

https://docs.yoctoproject.org/ref-manual/variables.html#term-BB_PRESSURE_MAX_CPU
but it also depends upon the build system configuration

maroueneboubakri, 2024-11-03:
> Glad, its working now, although -j1 and single thread is not ideal. Usually in latest yocto bitbake there is a pressure monitor which takes care of not overloading the system and adjusts issues new [...]

Yes indeed, that has been changed just to pass gcc build only.