riscv/meta-riscv #513: linux-beaglev-dev-6.9.0+git-r0 do_patch failed

kanakshilledar, 2025-03-15:
I am trying to build core-image-minimal for beaglev-ahead and the build is failing at do_patch step. Full log as follows:

``` console
$ BB_NUMBER_THREADS=6 bitbake core-image-minimal
Loading cache: 100% |                                                                                                                                                          | ETA:  --:--:--
Loaded 0 entries from dependency cache.
Parsing recipes: 100% |#########################################################################################################################################################| Time: 0:00:21
Parsing of 955 .bb files complete (0 cached, 955 parsed). 1922 targets, 101 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.9.1"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-24.04"
TARGET_SYS           = "riscv64-poky-linux"
MACHINE              = "beaglev-ahead"
DISTRO               = "poky"
DISTRO_VERSION       = "5.1"
TUNE_FEATURES        = "riscv64"
meta
meta-poky
meta-yocto-bsp       = "master:6610cad12a062592956257961a790ec6a3012b8b"
meta-riscv           = "master:42d2b51fa9b4f02a5fd522a7f6f2bfca508c3411"

Sstate summary: Wanted 1739 Local 0 Mirrors 0 Missed 1739 Current 0 (0% match, 0% complete)#####################################################################                | ETA:  0:00:00
Initialising tasks: 100% |######################################################################################################################################################| Time: 0:00:04
NOTE: Executing Tasks
ERROR: linux-beaglev-dev-6.9.0+git-r0 do_patch: Could not apply patches for beaglev-ahead.
ERROR: linux-beaglev-dev-6.9.0+git-r0 do_patch: Patch failures can be resolved in the linux source directory /home/kanak/yocto/build-beaglevahead/tmp/work-shared/beaglev-ahead/kernel-source)
ERROR: linux-beaglev-dev-6.9.0+git-r0 do_patch: Execution of '/home/kanak/yocto/build-beaglevahead/tmp/work/beaglev_ahead-poky-linux/linux-beaglev-dev/6.9.0+git/temp/run.do_patch.1016882' failed with exit code 1
ERROR: Logfile of failure stored in: /home/kanak/yocto/build-beaglevahead/tmp/work/beaglev_ahead-poky-linux/linux-beaglev-dev/6.9.0+git/temp/log.do_patch.1016882
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/home/kanak/yocto/sources/poky/meta/recipes-devtools/quilt/quilt-native_0.68.bb:do_populate_sysroot', '/home/kanak/yocto/sources/poky/meta/recipes-kernel/kern-tools/kern-tools-native_git.bb:do_populate_sysroot', 'virtual:native:/home/kanak/yocto/sources/poky/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: ['quilt-native', 'patch-native', 'attr-native']
| NOTE: Skipping as already exists in sysroot: ['gettext-minimal-native', 'cmake-native', 'libtool-native', 'texinfo-dummy-native', 'kern-tools-native', 'openssl-native', 'expat-native', 'ncurses-native', 'zlib-native', 'git-native', 'make-native', 'perl-native', 'bzip2-native', 'xz-native', 'zstd-native', 'ca-certificates-native', 'curl-native', 'gdbm-native']
| DEBUG: sed -e 's:^[^/]*/:/home/kanak/yocto/build-beaglevahead/tmp/work/beaglev_ahead-poky-linux/linux-beaglev-dev/6.9.0+git/recipe-sysroot-native/:g' /home/kanak/yocto/build-beaglevahead/tmp/sysroots-components/x86_64/quilt-native/fixmepath | xargs sed -i -e 's:FIXMESTAGINGDIRTARGET:/home/kanak/yocto/build-beaglevahead/tmp/work/beaglev_ahead-poky-linux/linux-beaglev-dev/6.9.0+git/recipe-sysroot:g; s:FIXMESTAGINGDIRHOST:/home/kanak/yocto/build-beaglevahead/tmp/work/beaglev_ahead-poky-linux/linux-beaglev-dev/6.9.0+git/recipe-sysroot-native:g' -e 's:FIXME_PSEUDO_SYSROOT:/home/kanak/yocto/build-beaglevahead/tmp/sysroots-components/x86_64/pseudo-native:g' -e 's:FIXME_HOSTTOOLS_DIR:/home/kanak/yocto/build-beaglevahead/tmp/hosttools:g' -e 's:FIXME_PKGDATA_DIR:/home/kanak/yocto/build-beaglevahead/tmp/pkgdata/beaglev-ahead:g' -e 's:FIXME_PSEUDO_LOCALSTATEDIR:/home/kanak/yocto/build-beaglevahead/tmp/work/beaglev_ahead-poky-linux/linux-beaglev-dev/6.9.0+git/pseudo/:g' -e 's:FIXME_LOGFIFO:/home/kanak/yocto/build-beaglevahead/tmp/work/beaglev_ahead-poky-linux/linux-beaglev-dev/6.9.0+git/temp/fifo.1016882:g'
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_patch
| (1/7) 0001-dt-bindings-reset-Document-th1520-reset-control.patch
| (2/7) 0002-reset-Add-th1520-reset-driver-support.patch
| (3/7) 0003-WIP-on-th_net-d3e1832c7d2d-riscv-mm-update-T-Head-me.patch
| [INFO]: check of .kernel-meta//patches//./0003-WIP-on-th_net-d3e1832c7d2d-riscv-mm-update-T-Head-me.patch with "git am" did not pass, trying reduced context.
| [INFO]: Context reduced git-am of .kernel-meta//patches//./0003-WIP-on-th_net-d3e1832c7d2d-riscv-mm-update-T-Head-me.patch with "git am" did not work, trying "apply".
| Context reduced to (2/2) to apply fragment at 37
| Context reduced to (1/1) to apply fragment at 58
| Context reduced to (2/2) to apply fragment at 230
| Context reduced to (2/2) to apply fragment at 282
| error: patch failed: arch/riscv/boot/dts/thead/th1520.dtsi:249
| error: arch/riscv/boot/dts/thead/th1520.dtsi: patch does not apply
| [ERROR]: Application of .kernel-meta//patches//./0003-WIP-on-th_net-d3e1832c7d2d-riscv-mm-update-T-Head-me.patch failed.
|          Patch needs to be refreshed. Sample resolution script:
|              .git/rebase-apply/resolve_rejects
| ERROR: Could not apply patches for beaglev-ahead.
| ERROR: Patch failures can be resolved in the linux source directory /home/kanak/yocto/build-beaglevahead/tmp/work-shared/beaglev-ahead/kernel-source)
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/kanak/yocto/sources/meta-riscv/recipes-kernel/linux/linux-beaglev-dev.bb:do_patch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2081 tasks of which 0 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/kanak/yocto/sources/meta-riscv/recipes-kernel/linux/linux-beaglev-dev.bb:do_patch
    log: /home/kanak/yocto/build-beaglevahead/tmp/work/beaglev_ahead-poky-linux/linux-beaglev-dev/6.9.0+git/temp/log.do_patch.1016882
Summary: There were 3 ERROR messages, returning a non-zero exit code.
```

Thanks
Kanak S

kraj, 2025-03-15:
Thanks for the report, this patch needs to be rebased on this kerel which is being built. if you can do that and submit a patch here, that will be awesome.

kanakshilledar, 2025-03-15:
sure, i will do it and file a PR, additionally i will bump the kernel versions to v6.13.

kanakshilledar, 2025-03-15:
doing some analysis of this patch, the changes which it is trying to implement are already done in a different [patch](https://github.com/torvalds/linux/commit/7e756671a664b73b2a3c0cc37fd25abf6bcd851e), thus making it unnecessary and safe to remove
i will check for others as well.

PR #514 (2025-03-16): linux-beaglev-dev: fix kernel do_patch and upgrade
linux-beaglev-dev recipe was failing at the patching step due to outdated patches, refined the patches
required for kernel to boot and upgrade the kernel version to 6.13, most of the patches are upstreamed in newer kernels and only a few patches are still
remaining to be merged.

Fixes: #513
closed 2025-03-16 by kraj