riscv/meta-riscv #528: Lack of Github Actions for this layer

kanakshilledar, 2025-04-23:
Hi,
I noticed this layer is lacking Github Actions or some form of CI/CD pipeline. Which will check for build failures,
DCO in commit messages, etc. Is there any specific reason for not implementing this? Or it is planned for the future?
We can add the builder similar to [meta-raspberrypi](https://github.com/agherzan/meta-raspberrypi).

This pipeline will be helpful as we have a bunch of patches in our board BSP and these will cause build failure whenever the patch gets upstreamed or some other change gets merged. This pipeline will help in identifying what board is failing and one can quickly resolve the issue.

kraj, 2025-04-23:
Its desired but we do not have resources. For meta-raspberrypi, we have sponsored runners and thats why we can do regular CI, I do local testing on my own for this layer at the moment, but I would certainly like to have GH Actions enabled CI

kanakshilledar, 2025-04-23:
Do we have any alternatives to get the runners apart from the Github self-hosted runners?
For yocto we do need to have substantial amount of compute and storage.

kraj, 2025-04-23:
> Do we have any alternatives to get the runners apart from the Github self-hosted runners? For yocto we do need to have substantial amount of compute and storage.

We need self-hosted runners for doing anything useful with yocto builds, the build times are simply too long for any free runner plans.

kanakshilledar, 2025-04-23:
Are there any future plans for getting self hosted runners? Or do we have to manage on our own, like you are doing local testing?

closed 2025-05-09 by kanakshilledar